c3d/db48x #13: Slow editing on battery

c3d, 2023-06-27:
Editing while on battery power is really slow

balta1980, 2023-06-28:
Also on battery functions like sin, log, square root crash the calc.
 On battery, even doing, nothing the battery drains very fast.

balta1980, 2023-06-29:
I figured out that if the calc did not crash on battery when run for example log just got freeze. When plugged to USB again the function finish correctly

balta1980, 2023-06-29:
I am in love with this program.

c3d, 2023-06-29:
> I figured out that if the calc did not crash on battery when run for example log just got freeze. When plugged to USB again the function finish correctly

Yes, this is consistent with my experience. My suspicion is that there is something wrong with how I use the timers to refresh the cursor. This is high on my to-do list.

c3d, 2023-06-29:
> Also on battery functions like sin, log, square root crash the calc.

Probably more like a hang, as you noted. This is #12 

> On battery, even doing, nothing the battery drains very fast.

Interesting. How do you evaluate / measure that?

c3d, 2023-07-08:
I believe that this was solved in [version 0.2](https://github.com/c3d/DB48X-on-DM42/releases/tag/v0.2). Please re-open if the problem still exists.

closed 2023-07-08 by c3d