c3d/db48x #62: Thousands separator when entering a big number

balta1980, 2023-07-01:
The businesses calculator HP17BII has a great feature, when you are entering a big number like 1000000, the prompt add automatically the set thousands separator perhaps "," or "." Would you please add the space you already use on v0.2

This feature is very helpful because you don't need to count zeros the hard way.

Thanks

balta1980, 2023-07-01:
I didn't realize you already develop on 0.2 several thousands separator, well, the idea is that whichever the user has set, it is shown when in the line edit is entering a big number. I understand this feature is not on hp48,49,50 nor prime, but very useful for business. Thank you.

c3d, 2023-07-05:
I will try to find a smart way to do it. It's not as easy as it looks, because unlike what happens with the HP17B, parsing of the command line normally does not happen before you hit the `ENTER` key

c3d, 2023-07-10:
This seems doable after all...

https://github.com/c3d/DB48X-on-DM42/assets/1695924/5c72f59f-92cb-4ab3-9c82-9399c36148f0

balta1980, 2023-07-10:
Wonderful, thank you very much for all you are doing

commit 193f1708 (2023-07-11): Dynamically maintain thousand separators while editing

Dynamically edit the position of the separators when editing a number.
This also addresses issues with the normal display of numbers.

Fixes: #62

Suggested-off-by: Baltazar J Diaz Z
closed 2023-07-11 by c3d