c3d/db48x #255: Implement ->NUM

c3d, 2023-07-31:
Implement a `->NUM` operation to be able to convert fractions to real numbers.

closed 2023-08-07 by c3d

spiff72, 2023-08-08:
Is this ->NUM function available as a shortcut on the KB?

lmamakos, 2023-08-24:
This still doesn't quite work completely to evaluate an expression.  So I do this sequence one the keyboard:
```
355
113
/
𝝅
-
``` 

and I'm left with `'355/113-𝝅'` on the stack.   When I try to invoke `➝Num`, I get a "Bad argument type" error reported.   

On my HP-48G, the `355 133 /` evaluates immediately, and subtracting `𝝅` from it can be then simplified to a number using `➝Num`.

c3d, 2023-08-24:
> Is this ->NUM function available as a shortcut on the KB?

The intent is for it to be mapped to the "EEX" key (which would be cycling between possible representations of a number). It's not there yet.

c3d, 2023-08-24:
> This still doesn't quite work completely to evaluate an expression. So I do this sequence one the keyboard:
> 
> ```
> 355
> 113
> /
> 𝝅
> -
> ```
> 
> and I'm left with `'355/113-𝝅'` on the stack. When I try to invoke `➝Num`, I get a "Bad argument type" error reported.

That's a bug. I filed #345.

 
> On my HP-48G, the `355 133 /` evaluates immediately, and subtracting `𝝅` from it can be then simplified to a number using `➝Num`.

There is a change in behaviour there between the HP48 and HP50. The HP48 does not have fractions or big numbers. The HP50 and DB48X do support this. So the HP50 behaves like DB48X in that respect (except for the bug in `->NUM` 😄 )

The HP50 also has a flag indicating if you want to immediately evaluate as numeric. This is not implemented yet. I filed #346.

BTW, when you find a problem like this, I encourage you to file an issue for each problem rather than mention it in a comment on some other issue. That ensures it will be addressed some day 😄  

> 
> I'm running version 0.4.2 on my DM42.

lmamakos, 2023-08-24:
Thanks!  I wasn't sure if you wanted to re-open this issue or not; I'll keep that in mind for next time.  Thanks for being so responsive and for the wonderful software.

commit c6919772 (2023-08-02): Implement →Num and →Q

Following several requests from various users, implement `→Num`
(converting numbers to decimal representation) and its counterpart,
`→Q` (converting numbers to fractions), including the special cases
for pi and i.

Fixes: #255
Fixes: #282
commit 4ef13cd0 (2023-08-03): Implement →Num and →Q

Following several requests from various users, implement `→Num`
(converting numbers to decimal representation) and its counterpart,
`→Q` (converting numbers to fractions), including the special cases
for pi and i.

Fixes: #255
Fixes: #282
commit 57c273b5 (2023-08-07): Implement →Num and →Q

Following several requests from various users, implement `→Num`
(converting numbers to decimal representation) and its counterpart,
`→Q` (converting numbers to fractions), including the special cases
for pi and i.

Fixes: #255
Fixes: #282