c3d/db48x #270: Argument of complex numbers computed in radian, irrespective of angle mode

c3d, 2023-08-01:
### Discussed in https://github.com/c3d/DB48X-on-DM42/discussions/174

<div type='discussions-op-text'>

<sup>Originally posted by **camillew** July 23, 2023</sup>
Hello,
I was trying to use complex numbers, and noticed that the ARG function (as well as the polar rendering) seems to always return a number in radians, irrespective of the angle mode.

In the same way, I see that the rendering of the polar mode changes (with the degree, rad, grad, npi suffixes), but the argument does not change.

However, the ATAN function seems to respect the angle mode.

In any case, than you for this nice project!
Camille
</div>

c3d, 2023-08-01:
@camillew Thanks. Will probably be fixed by https://github.com/c3d/DB48X-on-DM42/commit/1eae092e1d50288b439ebd12d5f8b1e8ae756e49

camillew, 2023-08-02:
Yes, this is fixed with this change, thank you!