c3d/db48x #271: Modes menu config issue

c3d, 2023-08-01:
### Discussed in https://github.com/c3d/DB48X-on-DM42/discussions/262

<div type='discussions-op-text'>

<sup>Originally posted by **camillew** July 31, 2023</sup>
Hello, 

I started testing the new release 4.0, and there seems to be a spurious ", ID_Modes);" at the end of the ModesMenu, resulting in a "Modes" entry at the very end of the menu, which drops a menu config on the stack when pressed.

Camille
</div>

c3d, 2023-08-01:
@camillew This is intentional. The `Modes` function returns a program that you can evaluate to reset the modes. The intent is something like this, where you execute the program P and if it changes settings, you restore them to what they were.

```
« Modes → UserProgram RestoreModes « 1 10 START UserProgram NEXT RestoreModes » »
```

camillew, 2023-08-01:
I see, thank you for the explanation!
And I see now that it is explained in the `settings.md` file...

closed 2023-11-12 by c3d