c3d/db48x #273: Implement object switching on the EEX key

c3d, 2023-08-01:
Implement a smart object switching on the EEX key:

- Switch between fractions and decimal representation
- Switch between polar and rectangular for complex numbers
- Switch between vector and and list form
- Switch units for unit objects

How should the feature be named?

camillew, 2023-08-02:
Maybe ModeCycle? Or Rep[resentation]Cyle?
polar<->rect and list<->vector would work as cycle, fraction<->decimal too (->Num ->Q) but what about unit objects?

Would it be specialized for some classical pairs, like (cm<->in, km<->mi, etc.)? But then, would fractions be allowed in unit objects? As in, if we have (1/4)_m, what would the switching do?

c3d, 2023-08-03:
`ModeCycle` suggests a mode change, but this is really a value change.
I prefer `RepCycle` or maybe `ObjCycle`.

Tp amswer the question about units, I was thinking this would switch
between nearby SI units, i.e.

1km -> 1000m
1000km -> 1_000_000m -> 1Mm

For 1/4_m, I would expect 25cm -> 0.25m, but then I think that the
question is if there would also be a cycle of fraction vs. decimal.

Need to think more about this.

c3d, 2023-10-08:
Finally settled on calling that function `Cycle`.

commit b42ebda2 (2023-10-08): Implement the `Cycle` function

The `Cycle` function is mostly intended for interactive use, mapped
on the `EEX` key.

Currently implemented cycles:

* Polar <-> Rectangular for complex numbers
* Decimal <-> Fraction
* Integer <-> Based (cycles through the bases)
* Array <-> List <-> Program
* Text <-> Symbol

Fixes: #273
closed 2023-10-08 by c3d