c3d/db48x #283: Shift actions in VARS menu swapped (compared to HP48)

camillew, 2023-08-02:
In the VARS menu, left shift allows to recall values, while right shift (double press) allows to store values. This is the opposite of the HP48 behavior. 

I was wondering whether the fact that we have a single shift button would justify a different approach than the HP48 one, but considering some typical use cases, I imagine it may make sense to swap the second and third menu rows: For variables containing values, EVAL and RCL are equivalent, so that having single-shift RCL is less useful than having single-shift STO.

Any concern with swapping the second and third rows of the VARS menu?

c3d, 2023-08-02:
Yes, this is a deliberate choice. Overwriting a value is a bit dangerous, so I prefer to have it on second row.

c3d, 2023-08-02:
That being said, just to be clear, the fact that this was deliberate and intentional does not mean it's the right choice. I must admit that I had not really considered the fact that you probably perform `STO` more often than `RCL`.

Let me mull over it a bit.

spiff72, 2023-08-08:
I actually agree that overwriting a variable makes sense to live on the 2nd-shift due to it being "dangerous".