c3d/db48x #288: Multiplying by -1 does nothing

camillew, 2023-08-03:
Running "-5 -1 *" returns -5. This seems to be due to the special casing of multiplication using the integer::is_one(), which is incorrect for neg_integer and would need to be specialized. I imagine that big_num may have a similar issue.

camillew, 2023-08-03:
This issue is resolved by specializing the `neg_*` types in `bool object::is_one(bool error) const`, starting with
```
{
    id ty = type();
    switch(ty)
    {
    case ID_neg_integer:
    case ID_neg_bignum:
    case ID_neg_fraction:
        return false;
...
```

c3d, 2023-08-03:
I have a fix ready, but it looks like you are quite ready to submit patches ;-)

camillew, 2023-08-04:
Yes, I would be happy to try to send patches, but I am not yet able to compile Qt in such a way that it works with the simulator, which in turns prevents me from running/enriching test cases. Hopefully I'll get there soon :)

commit 8e261592 (2023-08-06): Negative integers, fractions and bignums cannot be 'one'

This leads to `-1 * 3 == 3` due to abusive simplification.

Fixes: #288

Reported-by: Camille Wormser <cwormser@google.com>
Suggested-by: Camille Wormser <cwormser@google.com>
commit 59091a48 (2023-08-07): Negative integers, fractions and bignums cannot be 'one'

This leads to `-1 * 3 == 3` due to abusive simplification.

Fixes: #288

Reported-by: Camille Wormser <cwormser@google.com>
Suggested-by: Camille Wormser <cwormser@google.com>
closed 2023-08-07 by c3d