c3d/db48x #308: Need directory naviagation method and other directory enhancements

spiff72, 2023-08-08:
I see that the CRDIR and PGDIR functions have now been added, but I don't see an obvious way to navigate them.

I recall there being KB shortcuts (uh-oh!) on the 48S series?  The 48G has LeftShift UP (on the ' key).

Additionally, it might be helpful to have the little black bar (a few pixels wide) above the black background of the softkey for directory labels on the softkeys.  Otherwise I can't tell the difference between a directory label and a variable label...

spiff72, 2023-08-08:
...and it was staring me right in the face.  I even had the overlay installed!

The UP function is on the 2nd-shift backspace key!  Sorry about that.

However, I do still think my second comment about the black bar (or some other identifier - maybe similar to the menu one with the little triangle in the corner?) would be beneficial.

spiff72, 2023-08-08:
Tacking on one more thought for an enhancement:
Can there be an option in settings to allow for displaying current directory in the status bar?  This might take a lot of space, but the 48 series had a "breadcrumb" trail showing the directory location - e.g. [HOME TESTDIR]

Showing the whole trail might be excessive, but I think at least the current directory would be useful.

c3d, 2023-08-08:
My current thinking is that at some point, there will be a special variable (e.g. `Header`) that will be evaluated to put whatever you want in the header. So putting `PATH` in that variable will get the result you want.

spiff72, 2023-08-08:
Ok. I get confused when the variable softmenu disappears when u are in a directory where no variables are stored

c3d, 2023-08-16:
@spiff72 Is this something you could get used to, or would you prefer to have (at least an option for) empty menus in that case?

spiff72, 2023-08-16:
It would be nice to at least have the option of a persistent var menu, but i'm flexible

jthole, 2023-12-29:
Regarding having the current directory in the header, I like the idea of populating a Header variable with the 'Path'. But to save space, and in the interest of readability, I would not display the "HomeDirectory" part ... a blank area in the status bar would indicate that the user is in the home directory.