c3d/db48x #476: Random reset on 0.4.8

balta1980, 2023-10-26:
After a day of using 0.4.8 firmware:

[https://youtube.com/shorts/UOwYYszr4yU?si=w-_QFNlmRmH6zmhc](url)

After stopping using the calculator and powering off **for several minutes** and try to turn it on again it crashes and reboot.

Thank you.

c3d, 2023-10-26:
Thanks for sharing the video. That is really weird. Mine has been running for over a week and I did not see that problem.

Did you reflash the QSPI (not just the PGM file)? Having a mismatched QSPI / PGM pair is not detected, and it can lead to random crashes like that.

c3d, 2023-10-28:
Did you have the problem again?

I've been running 0.4.8 on battery and USB for nearly a week, restored it to factory settings, and I could not reproduce the problem.

balta1980, 2023-10-28:
Just in case, I re flash QSPI and program but, and after a day I got same behavior than before. If I power off calculator and wait around 5 seconds and try to turn it on it reboot.

balta1980, 2023-10-28:
Look at this new video [https://youtu.be/fW0FyyR8Stw?si=6y5BuLitEKg46MyS]()

c3d, 2023-10-28:
Hmmm... I think that I have a new explanation. You battery seems to be very low. It's showing 2.4V. I've never been so low. Could you try replacing the battery and see if that helps?

I need to see what the DM42 does in that scenario. Out of curiosity, could you, with the same battery, reload the DM42 program and see if it complains? Or if it behaves normally?

c3d, 2023-10-28:
In the other video, I can see the battery voltage dropping very quickly. It starts at 2.7V, but then it's down to 2.5 or less. Mine starts at 2.680, but then _increases_ to 2.702V or something like that.

I suspect that the loading of the off-image that occurs when you power off is actually somewhat CPU intensive, and might be draining an already dying battery.

c3d, 2023-10-28:
That being said, I find it suspect that the battery voltage is not drawn in gray, as it should if the DMCP `get_lowbat_state()` returned a low-bat state. Apparently, it does not.

I know that the C43 / C47 team had a lot of trouble getting the low-battery behaviour to be decent on this platform. See [this issue](https://gitlab.com/rpncalculators/wp43/-/issues/456) for an example.

c3d, 2023-10-28:
Apparently, the problem [also exists with the stock DM42 firmware](https://forum.swissmicros.com/viewtopic.php?t=3432)

balta1980, 2023-10-29:
After replacing the battery, no more problems. I am sorry for wasting your time. Thank you very much.

c3d, 2023-10-29:
You did not waste my time at all. On the contrary, this led to an improvement of battery icon rendering that will make the low-battery situation more visible I am also considering forcing the machine to 'off' with a message in that case. So very useful information.

commit 858659da (2023-10-29): ui: Display battery level more agressively

It looks like we can have a low-battery situation around 2.6V,
so gray out the voltage display in that case, and scale the
display of the battery icon more aggressively to not give a false
sense of security.

Fixes: #476
commit 72b8cbd7 (2023-11-01): ui: Display battery level more agressively

It looks like we can have a low-battery situation around 2.6V,
so gray out the voltage display in that case, and scale the
display of the battery icon more aggressively to not give a false
sense of security.

Fixes: #476
closed 2023-11-02 by c3d