c3d/db48x #550: Update units menu to reintroduce `cm` and other derived units

c3d, 2023-11-06:
After adding support for the units file, the menus were not reorganized to match what was there before.

Doing so would require the ability to add units that are shown only in menus, but not looked up as individual units. Units with a definition beginning with an `=` sign will only appear in menus, but will not be looked up.

c3d, 2023-11-06:
Also, @spiff72 reported that there is a bogus `miUS` menu entry in the built-in database.

spiff72, 2023-11-06:
Is there documentation of how to use the units file?  I see that there is an example in the repo, but is this a file that would be placed within the FAT folder structure, or is it read at compile time?

closed 2023-11-12 by c3d

c3d, 2024-03-27:
> Is there documentation of how to use the units file? I see that there is an example in the repo, but is this a file that would be placed within the FAT folder structure, or is it read at compile time?

It should be placed inside the `CONFIG` directory.

commit e974ad29 (2023-11-10): units: Reorganize menus both in units file and built-in units

Reorganize the units menu both in the units file and in the built-in units

Fixes: #550