c3d/db48x #561: Units Conversion not working correctly for Area and Volume (and possibly others)

spiff72, 2023-11-13:
I have done some experimenting with UNITS conversions, and I am getting some odd results.  I loaded a clean state after loading v0.4.11...

Within the length section, the conversions seem to work OK between meters and inches, yards, etc.

However, under the Area section, enter 1 m^2 from the F-keys, and then try to convert to any other unit on the first page (cm^2, in^2, ft^2 or yd^2) via the single shift F-keys, the result is unchanged (1 m^2 is displayed and there is no error message shown).

Similarly, in the Volume section, conversions from 1 m^3 to other units result in 1 m^3 (no error displayed).  Converting 1 m^3 to liters (lowercase L), or gal results in an inconsistent units error.

I also don't see consistency between the demo you did for the units from the recent youtube demo of v0.4.11 and compared to the example units.csv order.  Not sure if this is a symptom of the recursion bug and directories...but the "Quick" and "Money" menus are missing.

spiff72, 2023-11-13:
Quick update:
I just tried creating a folder on the USB drive containing a downloaded copy of the units.csv file (stored as /config/units.csv).

This solved the mismatch between the demo video and the units menu I am seeing here, but did NOT solve the issue of converting the non-length units like area and volume described above.

I checked this with a freshly cleared state, then re-flashed the QSPI and the pgm files, getting no errors from trying to restore the previous state with the subfolder errors.

c3d, 2023-11-14:
Thanks. I will take a look

spiff72, 2023-11-14:
> Thanks. I will take a look

I should add that there are a LOT of other units that either don't do anything when using the single-shift conversion method, and others that say inconsistent units (like mph and km/h).  Others will do the conversion, but still pop the inconsistent units error (like under the mass menu, converting lbs to oz).

c3d, 2023-11-14:
I was unable to reproduce this problem. It works fine for me, both on DM32, DM42 and simulator.

There must be some setting that I am overlooking. Could you share the state file with me?

c3d, 2023-11-14:
For the record, here are some of the tests I did:

- Convert 30 m^3 to liters gives me 30000l.
- 3.0 l to cm^3 gives me 3000 cm^3
- Converting that to galons gives me rougly 7520.82 gal
- Using the quick menu, 3 km/h converts to 15625/8382 mph
- 60.0 mph to km/h gives me 96.56 km/h.

So I think there must be somthing in the settings.

c3d, 2023-11-14:
Also, this time I tried with numeric and symbolic mode, both work.

c3d, 2023-11-14:
Can't resist using the new screen snapshot feature:
[image]

spiff72, 2023-11-14:
Yes - I will send it in a couple hours (at an appointment right now)…

spiff72, 2023-11-14:
I just emailed you my state file and my units.csv file (which I downloaded directly from this repo with no changes--at least as far as I know).

spiff72, 2023-11-16:
v0.4.12 seemed to fix this issue for me, so I am going to close this.

closed 2023-11-16 by spiff72