c3d/db48x #637: SQRT(-1) gives an error message

jthole, 2023-12-07:
Running version 0.5.1 on a DM42:

SQRT(-1) gives me the following error message: "SQRT error: Argument outside domain".

When I enter SQRT(-1 + 0i), I get the correct answer (0 + 1i).

Expected behavior would be, that entering SQRT(-1) gives me (0 + 1i) as result.

c3d, 2023-12-11:
This is an expected, but undocumented deviation from HP's RPL.

HP48's will automatically return complex values for real input

HP50 has a -103 flag for real vs. complex results, but the documentation about it is very poor, and it does not seem to impact the result of something like `sqrt(-1)`.

In DB48X, the intent is that a real function will stay in the real domain, under control of a flag that determines if you want complex or real results. Whether I map this flag to -103 or something else is a bit unclear at the moment. I recently added the flag in one of my dev branches, but it's not connected to anything yet.

c3d, 2024-06-27:
Fixed when implementing #951. This maps to flag -103 as indicated in previous comment. Closing.

closed 2024-06-27 by c3d