c3d/db48x #659: Unexpected behavior with applying units prefix to Hz unit (v0.6.0)

spiff72, 2024-01-02:
I haven't played around with my calc in a while, and I hadn't seen any release updates so I peeked this morning and saw that there were two!

Anyway - I saw in the changelog that unit prefix could be applied to Hz units so I gave it a try in v0.6.0 (I haven't tested in V0.5.2).

Steps to replicate:
enter 30e6 on the stack, and use the units menu to change it to 30e6 Hz
Use the double-shift CONV menu to apply the M (+6) prefix.

The resulting value on the stack is:
29.9998155011 MHz

Expected result would obviously be 30 MHz.

spiff72, 2024-01-02:
After another quick test, it seems there is something off in simple division operations:

Just entering 30e6 on the stack and dividing by 1000000 gives the same 29.9998155011 result?

c3d, 2024-01-02:
This appears to be a duplicate of #648, and works on my development branch.

The underlying issue is incorrect conversion from integer to decimal in some cases.

closed 2024-01-02 by c3d