c3d/db48x #671: Objects in hms unit not correctly entered

jthole, 2024-01-05:
v0.6.1 / sim

Enter on command line: 23:30:30_hms

Press Enter

First entry on the stack becomes 30:00:00 (instead of 23:30:30)

Pressing HMS-> returns 30 on the stack (should have been 23.508333)

c3d, 2024-01-18:
When I try to enter `23:30:30_hms` on the command line, I get a syntax error. This is expected: the `:` notation on the command line is for tagged objects. So I'm not sure exactly what you entered. I may also have changed something since 0.6.1 that causes it to correctly emit a syntax error in that case.

You can tag an object with a number label. For example `:23:12` on the command line tags the `12` object with label `23`, and it displays on the stack as `23:12`. Same if you do `23 12 ->Tag`.

I intend at some point to show the label on the stack with a different font or maybe grayed out, just to make sure this is not confused with something else. I had not realized there was also a confusion with the display of HMS objects.

c3d, 2024-01-18:
The user interface for HMS is not stable yet. I'm unhappy with how it works at the moment.

On HP48, HMS is just a change in the _value_ of decimal numbers. On DB48X, I am trying to have HMS be a display property, the internal value being the value you would get from `HMS->` on HP48.

But it does not work the way I want yet. For example, if you type `1.23` then `->HMS`, you would expect `1:23:00` to show up on the stack. Instead, at the moment, you get 1:13:48. That's because the HP48 would show `1.1348` for `1.23 ->HMS`. But this means that at the moment, entering an HMS value is a two-step process where you have to start with an `HMS->`, which is a bit counter-intuitive. For example, `1.2345 HMS-> ->HMS` will show `1:23:45` on the stack.

It's yucky, and needs to be redesigned.

c3d, 2024-05-24:
With the addition of the "sequence of dots" interface, this issue has essentially been solved.

Now to enter `23:30:30`, you can for example use `23.30.30` and then `->HMS`, and you get the expected results `23:30:30`.

You can also directly add with `HMS+`.

Closing the issue.

closed 2024-05-24 by c3d