c3d/db48x #726: Unexpected/confusing behavior of the 1.0->1. vs 1.0->1 option settings under SEPS menu

spiff72, 2024-01-16:
This is a tricky one to describe clearly, but I will give an example:

I just upgraded to 0.6.2 today, and I CANNOT confirm whether this issue was present in 0.6.1 or not.  

Imagine you have a super-simple program saved as a variable called 'INC20' (so that it is accessible from the variable as a soft-key):

`<< 0.2 + >>`

This works fine if the mode is set to 1.0->1. under the "SEPS" menu.  Running the program increments the first level of the stack by 0.2 as expected.

However, if you switch to the 1.0->1 mode option, running the program increments the stack by 0.2, UNLESS you have a zero on the stack.  In this case, the result is 02 instead of 0.2 

I recalled the program stored in 'INC20', the result is:

`<< 02 + >>`

Additionally, if you are in 1.0->1. mode, and enter some values like 0.2 0.5 0.42 on the stack, and then change to 1.0->1 mode, the values entered change to 02 05 042 and will revert back to the decimal entered values if the mode is switched back to the original values.

Is this expected behavior or a bug?  I'm assuming this is an integer/real number thing, but it confused me to look at one of these programs and see the 02 instead of 0.2 value.

c3d, 2024-01-18:
This is definitely a bug. It looks like the `.` is not printed in a case where it should be.

c3d, 2024-01-18:
Original title "Unexpected/confusing behavior of the 1.0->1. vs 1.0->1 option settings under SEPS menu".

I will shorten it to describe what I believe is happening.

c3d, 2024-01-18:
This was an off-by-one in the test checking if the value was less than 1.
Fixed, will be published with 0.6.3.

commit 1e222132 (2024-01-19): decimal: Fix display of 0.2 in NoTrailingDecimal mode

In `NoTrailingDecimal`, we were incorrectly removing the trailing `.`
even if we had to display one digit after that. The reason was an
off-by-one error testing the `realexp` variable, because all decimal
digits are on the right of the exponent, i.e. `0.2` has a a `realexp`
of 0 and not -1.

Fixes: #726
commit 80979bfb (2024-01-20): decimal: Fix display of 0.2 in NoTrailingDecimal mode

In `NoTrailingDecimal`, we were incorrectly removing the trailing `.`
even if we had to display one digit after that. The reason was an
off-by-one error testing the `realexp` variable, because all decimal
digits are on the right of the exponent, i.e. `0.2` has a a `realexp`
of 0 and not -1.

Fixes: #726
commit f5ae876d (2024-01-20): decimal: Fix display of 0.2 in NoTrailingDecimal mode

In `NoTrailingDecimal`, we were incorrectly removing the trailing `.`
even if we had to display one digit after that. The reason was an
off-by-one error testing the `realexp` variable, because all decimal
digits are on the right of the exponent, i.e. `0.2` has a a `realexp`
of 0 and not -1.

Fixes: #726
closed 2024-01-22 by c3d