c3d/db48x #817: Incorrect value for `LN(0.002)`

c3d, 2024-02-07:
0.002 LN returns -5.4947879... whereas the real answer is -6.2146080..

Reported by [Nigel on hpmuseum forum](https://www.hpmuseum.org/forum/thread-20157-post-183249.html#pid183249)

closed 2024-02-12 by c3d

achan001, 2024-02-12:
Hi, c3d

It is odd log was based from log1p, and without cancellation correction.
We normally base log1p from log, because on the log side, it is easily scaled.
```
log(m * 10^p) = log(m) + p * log(10)
```
RHS can be calculated very accurately, with error only of log(m)
Except for the constants, this is *exactly* [shiift_ln2(x,b) = x + b*ln2](https://www.hpmuseum.org/forum/thread-21258-post-183730.html#pid183730)

For log1p(x), we just approximate with 1 term taylor series:

log1p(x) ≈ log(y) - (y-1-x)/y      , where y = float(x+1)

Based it the other way (not recommended), we have:

log(y) ≈ log1p(x) + (y-1-x)/y      , where x = float(y-1)

The problem is this would cause massive cancellation if y is tiny.
If x = float(y-1) = -1, we have RHS = log1p(-1) + 0/y = -∞

achan001, 2024-02-12:
If we have good guess for x = log(m), we can improve it cubically.

f = (e^x-m)   , f' = f'' = ... = e^x
Halley correction = f/(f' - (f''/2) * (f/f')) = 2 * (e^x-m)/(e^x+m)

We force m between √.10 to √10, to keep exponential effect small.
```
lua> ; require'mapm'
lua> ; mapm.places(49)  -- 50 sig digits
lua> m = mapm.asin(1)   -- pi/2
lua> m:log()
4.5158270528945486472619522989488214357179467855509E-1
lua> _:round(15)        -- 16 sig digits guess
4.5158270528945490000000000000000000000000000000000E-1
lua> x = _
lua> y = x:exp()
lua> x - 2*(y-m)/(y+m)
4.5158270528945486472619522989488214357179467855512E-1
```

commit 23c1c251 (2024-02-11): decimal: Compute `ln(0.002)` correctly

The loop adjusting the integer part may need to be repeated.

Fixes: #817