c3d/db48x #834: No rule to make target 'recorder/recorder.h' when building the simulator

vkadlcik, 2024-02-29:
I'm following BUILD.md on a fully updated Fedora 39 x86_64 but the simulator doesn't build:

```
$ git clone https://github.com/c3d/DB48X-on-DM42.git
Cloning into 'DB48X-on-DM42'...
remote: Enumerating objects: 11550, done.
remote: Counting objects: 100% (4561/4561), done.
remote: Compressing objects: 100% (978/978), done.
remote: Total 11550 (delta 3707), reused 4359 (delta 3574), pack-reused 6989
Receiving objects: 100% (11550/11550), 36.58 MiB | 33.35 MiB/s, done.
Resolving deltas: 100% (8953/8953), done.
$ cd DB48X-on-DM42/
$ make sim
mkdir -p build/dm42/release
touch build/dm42/release/.exists
echo "#define DB48X_VERSION \"0.6.5\"" > build/dm42/release/version-0.6.5.h
cp build/dm42/release/version-0.6.5.h src/dmcp/version.h
cd sim; qmake db48x.pro -o db48x.mak CONFIG+=release
Info: creating stash file /root/DB48X-on-DM42/sim/.qmake.stash
WARNING: Failure to find: ../recorder/recorder.c
WARNING: Failure to find: ../recorder/recorder_ring.c
cp sim/gcc111libbid-Linux-x86_64.a sim/gcc111libbid.a
make: *** No rule to make target 'recorder/recorder.h', needed by 'recorder/config.h'.  Stop.
```

c3d, 2024-03-02:
Try `git submodule update --init --recursive` before building.

c3d, 2024-03-02:
Also, I recommend building the simulator with `make sim` from top level, as there are dependencies on shared files.

vkadlcik, 2024-03-04:
Ah, didn't notice, sorry for the noise, I'm closing this. Unfortunately I still can't build the simulator on Fedora, I'll file another issue...

closed 2024-03-04 by vkadlcik