c3d/db48x #853: ../src/settings.cc:153:15: error: call of overloaded ‘format(settings&, renderer&, const char [23], long long unsigned int)’ is ambiguous

vkadlcik, 2024-03-04:
I can't build the simulator on Fedora (tried 39 and 38, clean installs in a virtual machine):

```
# dnf -y install arm-none-eabi-gcc arm-none-eabi-gcc-cs-c++ arm-none-eabi-newlib gcc-c++ git make findutils qt-devel qt6-qtbase-devel qt6-qtdeclarative-devel freetype-devel
# git clone https://github.com/c3d/DB48X-on-DM42.git
# sed -i'' 's|git@github.com:c3d/|https://github.com/c3d/|' DB48X-on-DM42/.gitmodules
# cd DB48X-on-DM42/
# git submodule update --init --recursive
# make sim &>make_sim.log
# echo $?
2 
```

I'm attaching the whole log but I guess the key problem is this:
```
# grep error: make_sim.log.
../src/settings.cc:153:15: error: call of overloaded ‘format(settings&, renderer&, const char [23], long long unsigned int)’ is ambiguous
```

[make_sim.log](https://github.com/c3d/DB48X-on-DM42/files/14481249/make_sim.log)

c3d, 2024-03-13:
Thanks @vkadlcik for the report.

I am in the process of fixing the Linux build. There are many warnings that the new GCC picks up, e.g. about unused variables or various problems with C-style struct initialization. I will also address the problem you reported.

c3d, 2024-03-14:
Changed title (was "../src/settings.cc:153:15: error: call of overloaded ‘format(settings&, renderer&, const char [23], long long unsigned int)’ is ambiguous".

prismv, 2024-03-19:
Hi Christohphe, eventually you have recognized already my SwissMicros post. I tried again with stable branch, and then saw this open issue. Therefore it is already registered. For completeness also the show-stopper on Arch Linux with recent clang++ compiler:

```
> make CXX=/usr/bin/clang++ CC=/usr/bin/clang QMAKESPEC=linux-clang sim 2>&1 | grep error
../src/ids.tbl:836:1: error: call to 'format' is ambiguous
1 error generated.

> LC_ALL=C clang++ --version
clang version 17.0.6
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/sbin

```

closed 2024-03-24 by c3d

JohnRobson, 2024-06-26:
> Thanks @vkadlcik for the report.
> 
> I am in the process of fixing the Linux build. There are many warnings that the new GCC picks up, e.g. about unused variables or various problems with C-style struct initialization. I will also address the problem you reported.

Please, do that, I want to compile the Simulator to share it. Thanks.

commit 483a5919 (2024-03-15): linux: Remove warning about fall-through switch statement

With gcc (GCC) 13.2.1 20231011 (Red Hat 13.2.1-4), there is a message about
fall-through switch statements. Add comment to silence the warning when the
fall-through is intentional.

There was one case where the fall-through was not intentional.
Missing `break` added.

Fixes: #853
commit d974b87b (2024-03-15): settings: Avoid build error due to bad `ularge` overload

Since `ularge` is defined as `uint64_t`, which is `unsigned long` on
Linux, but `unsigned long long` on macOS, so we need to be careful
about not using `1ULL` in constants, otherwise the `format` function
does not match.

Compiler error:

```
../src/settings.cc:153:15: error: call of overloaded ‘format(settings&, renderer&, const char [23], long long unsigned int)’ is ambiguous
```

Fixes: #853
commit f06d17af (2024-03-15): linux: Remove warnings about unused arguments

On Fedora 38, GCC incorrectly warns about unused arguments despite the
`__attribute__((unused)))` that should silence the warning.

Observed with gcc (GCC) 13.2.1 20231011 (Red Hat 13.2.1-4).

Fixes: #853