c3d/db48x #895: Windows emulator

super-tomcat, 2024-04-15:
Supporting windows with the emulator would be good (especially instead of the overpriced Mac!) since its also the most widely used operating system out there :-)

c3d, 2024-04-15:
If "overpriced Mac" is your problem, please use Linux :-) It's supported. There
is even a Linux subsystem for Windows.

As for supporting Windows, I'm pretty sure it's already there. The code is written against Qt, which runs on Windows. It's just a matter of someone with a willingness to build it.

Are you volunteering?

super-tomcat, 2024-04-15:
Unfortunately, i wont fill part of my windows drive with that linux subsystem (i tried it a long time ago), it is not worth it to me :-)
as for volunteering, well i dont code in C++, never have and never will

c3d, 2024-04-15:
OK, I tried a quick cross-build with Fedora mingw qt, and I was not too successful. I will return to it later.

super-tomcat, 2024-04-15:
okay, thanks for trying, and good luck with the whole project, i will keep an eye on it 👍