c3d/db48x #922: Graphical glitch in status bar

spiff72, 2024-05-07:
I am finding that with version 0.7.4, if I press the VAR key to show the variable menu, there is a graphical glitch present in the status bar on the far left.  It seems to distort the first part of the current date.  See attached photo (I don't know if it shows up in a screenshot because I can't recall the method of creating one).  I do not know if this glitch was present in previous versions.

Additionally, this only occurs when the stack is full enough to overflow at least one item off screen.  If there are no stack items above the uppermost visible stack level, the glitch doesn't appear.  Oddly, it also doesn't seem to appear when there is overflow AND the version command output is present at the first level.  If the output of the VERSION command is dropped (and there is still overflow), the glitch comes back.

Pressing EXIT to hide the VAR menu at the bottom clears up the glitch.  Pressing VAR brings it back.
[image]

c3d, 2024-05-07:
Ah, good catch!

What you see is the stack level, level 5 in your photo, you can somewhat make the 5 if you look carefully. This explains why you only see it if the stack is deep enough, or if the content of the stack (e.g. the result of `VERSION`) makes it so that the level number is not seen.

c3d, 2024-05-07:
I have a fix, it will be in 0.7.6. Thanks for reporting.

spiff72, 2024-05-07:
Ah - so I am seeing a black 5 superimposed over the light 5 (part of the date)?

commit 5d26c2f1 (2024-05-11): stack: Correctly set clipping for stack index

When drawing the stack index, the code was incorrectly resetting the
clipping area before drawing the stack index.

Setting the clipping area correctly in order to ensure we don't
corrupt the content of the header.

Fixes: #922
closed 2024-05-13 by c3d