c3d/db48x #953: Constants on the stack are corrupted after opening nonexisting help page

mahi2003, 2024-05-29:
1. Put equations with constants `π` (pi) and `e` (Euler's number) on the stack:
```
'2+π'
'1+e'
```
2. Open a non existing help page (for example hold a key like [1/x]). Once the "no help for..." error message appears the stack has changed to:
```
'2+Math'
'1+π'
```
`π` was changed to `Math` and `e` replaced with `π`!

3. Press the down button to edit the item in stack level 1, but cancel with [EXIT]. The original stack is restored.

It is however not just a visualization issue. If you execute [→Num] after step 2, stack level 1 becomes 4.1415 (1+pi)... whereas it should have been 3.7182... (1+e).

c3d, 2024-05-29:
This is a duplicate of #943, and is already fixed on the `dev` branch.

c3d, 2024-05-29:
Original user report:

https://www.hpmuseum.org/forum/thread-20157-post-187618.html#pid187618

mahi2003, 2024-06-03:
Confirmed fixed in 0.7.7!