c3d/db48x #955: Negative fraction may display 2 minus signs

mahi2003, 2024-05-29:
When the stack is in graphical mode, a negative fraction like `-1/2` is wrongly displayed with 2 minus signs:
```
  -1
- ——
   2
```
The same as an equation with division, `'-1÷2'`, is displayed correctly:
```
-1
——
 2
```

c3d, 2024-05-29:
Original user report is here

https://www.hpmuseum.org/forum/thread-20157-post-187618.html#pid187618

mahi2003, 2024-06-03:
Confirmed fixed in 0.7.7!

c3d, 2024-08-09:
There is a [comment on YouTube](https://www.youtube.com/watch?v=5iSjn6pGhqA&lc=UgwKLYdnhXYaOWMFPQp4AaABAg) that suggests this still exists on 0.7.12. Closing this one, I will let the original poster reopen or create new issue.

closed 2024-08-09 by c3d

c3d, 2024-08-11:
User confirmed he was on 0.7.4, i.e. prior to the fix being applied.