c3d/db48x #959: `xroot` does not work correctly with negative roots in algebraic entry

mahi2003, 2024-05-30:
When entering equation `'xroot(y,−x)'` it is put onto the stack as `'−xroot(y,x)'` which is obviously not the same. It's not just a visualization bug - the result is really wrong.

The same operation works fine in RPN entry: `y` `x` `+/-` `x√y` → `'xroot(y,−x)'`

mahi2003, 2024-05-31:
By the way, it just occurred to me that in algebraic entry the arguments of `xroot` are swapped compared to the HP 48/50...

DB48/50X: `xroot(radicand;index)`
HP48/50: `xroot(index,radicand)`

In RPN entry, both behave the same: Radicand in level 2 and index in level 1

c3d, 2024-06-09:
So after checking a number of functions, it looks like `xroot` is the only function where the order in algebraic mode is "backwards" compared to the order in RPN mode.

For example, `X Y MIN` results in `MIN(X,Y)`; `X Y RND` gives `RND(X, Y)`; `A B TRNC` gives `TRNC(A;B)`.

However, `A B XROOT` gives `XROOT(B;A)` on HP48. Currently it is `XROOT(A;B)` on DB48X.

This is also the only function I could find where there is a different stack diagram given in the _HP50G Advanced Reference_ where the order is different between the _Input/Output (RPN)_ and the _Input/Output (ALG)_ sections. A quick search in the manual shows no other occurence of "(ALG)" or "(RPN)" anywhere else in the document.

And indeed, when running in algebraic mode, `XROOT(A;B)` does result in the same thing as the DB48X version.

mahi2003, 2024-06-12:
I just noticed the Σ function (and possibly others) suffers from the same issue (minus sign being pulled out of the function in algebraic entry):

`'Σ(X;1;10;-X)'` → `'-Σ(X;1;10;X)'`
`'Σ(X;1;-10;X)'` → `'-Σ(X;1;10;X)'`
`'Σ(X;-1;10;X)'` → `'-Σ(X;1;10;X)'`
`'Σ(X;-1;10;-X)'` → `'Σ(X;1;10;X)'`
`'Σ(X;-1;-10;-X)'` → `'-Σ(X;1;10;X)'`

A special case is:

`'Σ(-X;1;10;-X)'` → `'Σ(-X;1;10;X)'`

This time the minus sign is not pulled outside the function (for `xroot`the issue did not occur with the first argument either), but does this even make sense? Shouldn't this throw a syntax or bad argument error? It does so in RPN entry. Stack:

```
'-X' 
1
10
'X'
```
`Σ` → Bad argument type

mahi2003, 2024-06-17:
Confirmed fixed in 0.7.8!

:heavy_check_mark: Argument order of `xroot` is now the same as HP48/50 (debatable which is right, but this way is less confusing for HP users).
:heavy_check_mark: Minus sign in algebraic entry `'xroot(y;−x)'` and `'Σ(X;1;10;-X)'` is preserved.
:heavy_check_mark: `'Σ(-X;1;10;X)'` in algebraic entry is no longer accepted (error on variable -X).

commit 5c6d3ed1 (2024-06-14): expressions: Parse negation after parentheses

When parsing `'xroot(a;-b)'` we were incorrectly returning
'-xroot(a;b)'` because we were treating `-` specially and before
treating parentheses or arguments.

Fixes: #959 (first comment)

Reported-by: mahi2003@users.github.com
commit 761c4216 (2024-06-14): xroot: Reverse `xroot` arguments in algebraic parsing

The `xroot` command is a special case on HP50, in that `A B XROOT`
returns `XROOT(B;A)` and not `XROOT(A;B)` like for other commands.

Special-case the parsing and rendering of that command.

(This is super-ugly, all the more so that the HP50G advanced user
manual actually indicates the order depends on algebraic vs. RPN mode)

Fixes: #959

Reported-by: mahi2003@users.github.com
commit 89b3312a (2024-06-14): sum: Do not accept a non-name in a sum or product

Parsing `'Σ(-X;1;10;X)'` was not failing at parsing time, since we are
simply parsing all arguments as expressions.

It's more user-friendly to special case this scenario and fail if we
don't have a name in the first position. The same also applies to
products.

Fixes: #959 (4th comment)
closed 2024-06-17 by c3d