c3d/db48x #960: Electrical units are under menu `Eng` instead of `Elec`

mahi2003, 2024-05-30:
Electrical units like `_V` and `_A` are under a menu called `Eng`. Maybe this is intentional, but I think it makes more sense to call the menu `Elec` like on the HP calculators.

c3d, 2024-06-07:
Interesting. Apparently, it is correct [in the built-in units][1], but the `units.csv` file does not have the "Elec" menu entry. I believe that this happened when I removed the "Visc" menu that had only two entries to make a bit of room (in order to have less than 18 categories so that you don't need to reach the "next" menu for some of them).

Clearly inconsistent between the built-in and CSV file, plus now I have one spare entry.

[1]: https://github.com/c3d/db48x/blob/05f60c078c0076bcf3b75bdedc92ea3bbaa187e4/src/unit.cc#L594

mahi2003, 2024-06-17:
Confirmed fixed in 0.7.8!

commit 66c460fa (2024-06-14): units: Split `Eng` to `Elec` and `Visc`

In order to better match the unit layout on the HP50, rename the `Eng`
menu to `Elec`, and move `P` and `St` units to the `Visc` menu.

Fixes: #960
closed 2024-06-17 by c3d