c3d/db48x #962: `mod` does not work correctly in algebraic entry

mahi2003, 2024-05-31:
The `mod` (modulo) function is implemented as an operator in algebraic entry, but it does not work correctly:

`'` `X` `mod` `Y` `ENTER` → `'XmodY'` (interpret as variable name _XmodY_ instead of mathematical operation)
`'` `9` `mod` `3` `ENTER` → `'mod3'` (9 is seemingly lost, but when pressing `EVAL` it is mysteriously put on stack level 2)

When you put a space before and after `mod`, it works correctly. Maybe DB48/50X should automatically insert spaces when pressing `mod` in algebraic entry?

mahi2003, 2024-06-01:
Duplicate of #906

mahi2003, 2024-06-01:
This may be a duplicate of #906, but I'm not sure. #906 seems to be about visualization while this is about entry in algebraic mode.

c3d, 2024-06-07:
Hi @mahi2003, could you please try again with 0.7.7?

Issue #906 is indeed about visualization, but with the result that it actually also solves that particular problem (at least for me).

c3d, 2024-06-07:
Oh, @mahi2003I think that I see what you mean. The problem is if you type `mod` with the menu entry, not if you manually type it.

Something like this key sequence (using letters for simplicity)

```
F X SHIFT SHIFT L F3 Y
```

This indeed does not insert spaces around `mod`.

mahi2003, 2024-06-17:
Confirmed fixed in 0.7.8!

commit a77f01d6 (2024-06-14): ui: Fix insertion/editing of infix with alphabetic names

When inserting `mod` or `and` in algebraic mode, make sure that we
insert or show displays around it.

Fixes: #962
closed 2024-06-17 by c3d