c3d/db48x #976: `ConvertToUnitPrefix` is very slow on real calculator

c3d, 2024-06-15:
Running `ConvertToUnitPrefix` for the value corresponding to the power generated for 3kg of matter takes several seconds on a real calculator.

mahi2003, 2024-06-20:
A random grab of units (on USB power):

`1_F` ~40 s
`1_V` ~15 s
`1_P` ~8 s
`1_bar` ~8 s
`1_W` ~7 s
`1_cal` ~4 s
`1_N` ~3 s
`1_mph` ~2 s
`1_l` ~1.5 s
`1_ha` ~1 s
`1_t` ~1 s
`1_s` ~0.5 s
`1_m` ~0.2 s
`1_USD` ~0 s

c3d, 2024-06-22:
I tracked it back to expression simplification that was added in 0.7.8, and applies to every single unit operation while factoring out units, making the factoring super slow.

Presumably fixed on `dev` now.

c3d, 2024-06-25:
Expression simplification was still happening for the evaluation of the definition of a unit.

commit 5c551ce1 (2024-06-22): units: Add `factoring` variable to limit simplifications

When factoring out units, we don't need to run through each
simplification in `expression::simplify`. We should also not lookup
names as symbols, but just let symbols as is.

Fixes: #988
Fixes: #976
commit f06c8240 (2024-06-27): units: Disable simplification during unit definition evaluation

When evaluating the definition of a unit, we disable simplifications,
which are not necessary if the unit was defined "correctly".

The evaluation of units as required for the equation simplification
loop was very expensive, because each run required going through the
`units.csv` file in order to check for units.

Fixes: #976
closed 2024-06-30 by c3d