c3d/db48x #979: Shift+Up does not return to beginning of entry

mahi2003, 2024-06-16:
_Down and Up are the buttons as labeled on the stock DM42. With the db48/50x overlay they are Left and Right._

In entry mode you can press Shift+Down to move the cursor to the next line or the last position in the line if no lines follow. Shift+Up moves the cursor up to the previous line, but if you are on the first (or only) line, Shift+Up does not seem to have any effect. I'd expect it to set the cursor to the first position.

`"Hello world"`

Set cursor to anywhere but first or last character. Press Shift+Down and the cursor moves to the last position. Now press Shift+Up and nothing happens.

```
"Hello
world"
```

With the cursor at the first position, press Shift+Down. The cursor moves to the second line. Press Shift+Down again and the cursor moves to the end of the second line. Press the Up arrow once (*). Now press Shift+Up to return to the first line. However, pressing Shift+Up once more does not return us to the beginning of the first line.

(*) If you don't do this, the cursor moves directly to the first position of the first line after pressing Shift+Up. It seems that pressing Shift+Up/Down does not update the last column the cursor is on.

mahi2003, 2024-06-16:
By the way, sometimes when writing a program or longer entry, I might accidentally press Shift+Shift+Up instead of Shift+Up to move up a line. The result is that I get a previous entry from history that overwrites the current entry (in other words my entire entry is lost). This can be quite frustrating...

I'm not sure to file this as a bug because technically it isn't - the overlay clearly states Shift+Shift+Up is history, but it's an easy mistake to make because you need Shift and Shift+Shift so often. Perhaps history should work only with the cursor at the first position? Or allow you to exit history without losing the current entry?

c3d, 2024-06-16:
Thanks a lot for the bug report regarding cursor movement. I will fix it right
away.

> By the way, sometimes when writing a program or longer entry, I might
> accidentally press Shift+Shift+Up instead of Shift+Up to move up a line. The
> result is that I get a previous entry from history [...]

> I'm not sure to file this as a bug because technically it isn't - the overlay
> clearly states Shift+Shift+Up is history, but it's an easy mistake to make
> because you need Shift and Shift+Shift so [...]

The good news is that the entry is not lost. Simply use Shift-Shift-Up again to go back further in the editing history. It's at most 8 deep, so in the worst case, you have to go back 8 levels.

I'll try to think of a convenient way to go through history without having to use double-shift every time. Maybe `HIST` could be a mode where up and down go through history, and you exit it with `ENTER`?

mahi2003, 2024-06-17:
Confirmed fixed in 0.7.8!

> The good news is that the entry is not lost. Simply use Shift-Shift-Up again to go back further in the editing history. It's at most 8 deep, so in the worst case, you have to go back 8 levels.

I had not noticed that. Thanks!

> I'll try to think of a convenient way to go through history without having to use double-shift every time. Maybe HIST could be a mode where up and down go through history, and you exit it with ENTER?

The history feature confuses me. Perhaps I don't fully understand its purpose/working.

`1` `ENTER` `2` `ENTER` `3` `ENTER` `4` `ENTER` `5` `ENTER` `6` `ENTER` `7` `ENTER` `8` `ENTER`

`HIST` (Shift+Shift+Up) → `8`
`HIST` → `7`
`HIST` → `6`
`EXIT`
`HIST` → `6`: Why doesn't it start from the beginning (8) again?
`HIST` → `8`
`HIST` → `7`
`HIST` → `1`: Huh?
`HIST` → `5`
`HIST` → `4`
`HIST` → `3`
`HIST` → `2`
`HIST` → `6`: Err...

commit f3ceb334 (2024-06-16): editor: Shuft-up returns to beginning of line if on first row

When on the first row of the editor and using the `up` key, set the
cursor to position 0.

Fixes: #979
closed 2024-06-17 by c3d