c3d/db48x #983: Entering polar value with different angle mode does not always work correctly

mahi2003, 2024-06-18:
I just watched the 0.7.8 demo on Youtube. One of the new features is the ability to enter a polar value with a different angle mode than currently set. This does not seem to work in all cases.

_< represents the angle symbol - I did not find the correct character_

With the calculator in `Grads` mode:

`1` `<` `45` `Deg` `ENTER` → :x: I expect `1<50g` on the stack, but instead I get `1<45g`. It's as if the `Deg` action was ignored even though the entry looked like `1<45°`prior to hitting ENTER.

The other way around works fine, though. With the calculator in `Deg`(rees) mode:

`1` `<` `50` `Grads` `ENTER`  → :heavy_check_mark: `1<45°` is put on the stack.

mahi2003, 2024-06-18:
Update: Further testing shows that only `Deg` is affected.

Entering polar values with `Grad`, `Rad` and `n×π` while the current angle mode is something else, works fine.

c3d, 2024-06-30:
Still reproducing, fix is now under test.

closed 2024-06-30 by c3d

mahi2003, 2024-07-01:
✔️ Confirmed fixed in 0.7.9!

commit 5c1c44fc (2024-06-30): parser: Parse degrees as an angle unit in phasors

Degrees can be either an angle marker in phasors, but also part of a
DMS notation. So parsing `10°` results in a DMS value for 10 degrees,
which is 10. As a result, the phasor parser was not seeing the degree
sign and interpreted the result as not having a unit.

Add a `separator` field to the `parser` class to record that we are
inside a phasor, so that the `integer` parser can avoid entering DMS
mode.

Fixes: #983
commit 8e9834cc (2024-06-30): tests: Add test for polar angle conversion.

When we enter a phasor with explicit degrees, we need to detect the
angle unit and apply it.

However, the angle unit is incorrectly parsed as a DMS value.
Detect that situation and report it.

Fixes: #983