c3d/db48x #988: Recalling value of `c` with `m` set to `1_kg` gives bad value

c3d, 2024-06-19:
When you recall the numerical value of `c` with `m` set to a value with a unit, that value takes precedence and you get something like `299792548_(1/100_kg)/s` on the stack instead of the actual unit.

c3d, 2024-06-19:
Same result whether you convert with `->Num` or directly recall the value from the constants menu.

mahi2003, 2024-06-20:
Interestingly, the issue resolved itself after resetting the calculator (including recalling c and `→Num`). Doing so also fixed #987 for me. Memory corruption?

mahi2003, 2024-06-20:
I found what starts breaking the units: Variables with the same name as a unit!

1. Reset calculator
3. `1` `m/s` → `1_m/s` ✔️ Step is optional, but confirms that at this time DB48/50X operates correctly
4. `10_g` `'m'` `STO`
5. `1` `m/s` → :x: `1_(10_g)/s`

If you delete variable m, the issue is gone again:

6. `'m'` `Purge`
7. `1` `m/s` → `1_m/s` ✔️

DB48/50X is substituting units with variables: 1_(<sub>unit</sub>m)/(<sub>unit</sub>s) becomes 1_(<sub>variable</sub>m)/(<sub>unit</sub>s) when variable m exists.

Note that it is not doing that when there are no operations on the unit: `1_m` does not become `1_(10_g).

I don't know whether this bug is the same as #987 and the _"No solution"_ issue you noticed in the 0.7.8 demo video at 19:15-19:35, but both seem to be related to units, so there is a good chance...

mahi2003, 2024-06-20:
**Update:** I just realized you already knew all that. I misread your first sentence. I feel so stupid now...

c3d, 2024-06-25:
Duplicate of #969

commit 5c551ce1 (2024-06-22): units: Add `factoring` variable to limit simplifications

When factoring out units, we don't need to run through each
simplification in `expression::simplify`. We should also not lookup
names as symbols, but just let symbols as is.

Fixes: #988
Fixes: #976
closed 2024-06-27 by c3d