c3d/db48x #990: `sqrt` and `xroot` do not work on values with units

mahi2003, 2024-06-20:
`2_m` `x²` → `4_m²` ✔️
`4_m²` `√x` → :x: Bad argument type
`8_m³` `3` `x√y` → :x: Bad argument type (and input values lost)

I also noticed that the HP 48/50 can apply functions like `FLOOR`, `RND`, `MIN`, etc to values with units whereas DB48/50X always returns bad argument type.

Interesting is `mod`. In DB48/50X:

`13_m` `6` `mod` → `'(13_m) mod 6'`

It does not appear to be possible to evaluate this (`→Num` doesn't work either). The HP 48/50 returns a bad argument type error in this scenario.

c3d, 2024-06-22:
Thanks a lot for reporting that, @mahi2003. I will try to implement it for the next release. It's not very complicated.

c3d, 2024-06-27:
Problem presumably also exists for power and cube root.

closed 2024-06-30 by c3d

mahi2003, 2024-07-01:
✔️ Confirmed fixed in 0.7.9 for `sqrt`, `cbrt` and `xroot`!

Shouldn't it be implemented for `FLOOR`, `RND`, `MIN`, etc like on the HP 48/50?

commit 2e6e5982 (2024-06-27): units: Accept units for `sqrt`, `cbrt` and `xroot`

Add processing of units for `sqrt`, `cbrt` and `xroot`, by applying
the exponent to the unit expression.

Also ensure that if we have `km^1` in a unit, we simplify it as `km`.

Also avoid a type error for `km^3` from `is_negative` being applied to
`km`.

Fixes: #990
commit 88a8f21b (2024-06-27): units: Accept units for `sqrt`, `cbrt` and `xroot`

Add processing of units for `sqrt`, `cbrt` and `xroot`, by applying
the exponent to the unit expression.

Also ensure that if we have `km^1` in a unit, we simplify it as `km`.

Also avoid a type error for `km^3` from `is_negative` being applied to
`km`.

Fixes: #990
commit 87d52b56 (2024-06-30): units: Accept units for `sqrt`, `cbrt` and `xroot`

Add processing of units for `sqrt`, `cbrt` and `xroot`, by applying
the exponent to the unit expression.

Also ensure that if we have `km^1` in a unit, we simplify it as `km`.

Also avoid a type error for `km^3` from `is_negative` being applied to
`km`.

Fixes: #990