c3d/db48x #991: Right shift indicator stays lit after triple pressing shift key rapidly

mahi2003, 2024-06-20:
When you triple press the shift key rapidly, the right shift indicator stays lit. It may take a few tries, but it's really easy to reproduce. If you have a multi-line menu open, the calculator correctly highlights the bottom row and pressing a menu soft key executes the bottom row one (in other words, the shift indicator was wrong).

c3d, 2024-06-22:
Thanks for reporting, @mahi2003. Reproduced and confirmed.

c3d, 2024-06-25:
Fix implemented. What matters is not pressing shift rapidly, but that no other refresh event for the header area happens after clearing the shift & alpha indicators.

The reason was that the screen refresh area for no indicators was empty, so the `refresh_dirty` function was doing nothing.

closed 2024-06-30 by c3d

mahi2003, 2024-07-01:
✔️ Confirmed fixed in 0.7.9!

commit d67deecd (2024-06-27): ui: Create a non-empty dirty rectangle when clearing annunciators

When annunciators where cleared, the horizontal coordinate for the
dirty rectangle could end up empty, causing the next call to
`refresh_dirty` to do nothing.

The manifestation was that if clearing the shift indicator and nothing
else refreshed the header rows, the shift indicator would remain lit.

It's not entirely clear why the simulator behaved differently in that
scenario. It is likely due to the way timer interrupts are simulated,
which caused a faster refresh, making it almost impossible to see.

Fixes: #991