c3d/db48x #992: Screen refresh broken?

mahi2003, 2024-06-20:
_I know the calculator slows down screen refreshes after inactivity — power savings. Initially it updates the screen every second, after a short while every 3 seconds, somewhat later every 10 seconds and then only once a minute._

_I think it's supposed to reset the refresh timer to update every second after a key press, but that does not seem to work in all cases. In fact, it seems that quite often pressing a key can disable the screen refresh entirely until the next key press._

1. Turn on seconds in the status bar. I do not recommend that, but it gives much faster feedback of screen updates.

2. While the clock is ticking, press the Up arrow button. The clock freezes (if it doesn't, try again). You can wait for minutes, it stays frozen. Inserting or removing the USB cable does not update the battery/voltage either. There's nothing special about the Up button - you can reproduce it with other buttons - but the Up button does not yet have a function assigned when on the stack, so it is very consistent.

3. Press `1`. The clock starts ticking again. If you wait refreshes will slow down again, but you don't have to wait that long.

4. Press `ENTER`. The clock is frozen again.

5. Press and hold `EXIT` for about a second. You may have to try a few times, but eventually the clock will start ticking again.

If you play a bit with the calculator, you'll find that it's easier to freeze the clock than to keep it running.

Note that entry mode seems to be immune for the issue.

One could argue that the calculator does not really need screen updates outside entry mode, but that renders the clock a bit useless (even without seconds) and it's a bit annoying the battery/voltage status does not update when inserting/removing the USB cable.

c3d, 2024-06-22:
Thanks for the report, @mahi2003.

Screen refresh is a delicate balance between avoiding consuming power when not needed and providing useful feedback.

Refreshing when USB power changes state is a good idea. I'll try to find a way to do that. That being said, I am not sure if the calculator even wakes up when the power changes state and nothing else does. I'll need to check that.

The "Up" scenario is interesting. You hear a beep when you do that, indicating that there is no action associated with the key in that context. I think that there is a path where the refresh timer is not restarted in that case.

This appears somewhat independent from the right shift refresh issue you reported in #991, but the two might be indirectly related. I'll tell you more when I investigate in depth.

c3d, 2024-06-23:
The problem does not reproduce on simulator, so there is a discrepancy between the simulator's implementation of timers and what happens on DMCP.

closed 2024-06-30 by c3d

mahi2003, 2024-07-01:
✔️ Confirmed fixed in 0.7.9!

commit feb28a38 (2024-06-27): ui: Compute the correct refresh rate even when nothing is displayed

When we display nothing, we go through `ui.draw_header`, which may not
need to draw time or date. In that case, we were not correctly
computing the refresh time (i.e. calling `draw_refresh`.

If nothing else was drawn (e.g. no annunciator), then the refresh time
would remain super high, causing the display refresh to freeze.

A similar problem exited for the battery display, and was also fixed.
In the process, added the `BatteryRefresh` setting, defaulting to 3s,
to indicate how often we should check the battery voltage.

Fix [... cut]