c3d/db48x #996: Usability - using equations

guirossibrum, 2024-06-25:
Hello Christophe, running v0.7.8 I am trying to add an equation to the library, I see you suggest to load the config:equations.csv file, edit and then store again. 
I tried to type the following '"config:equations.csv"RCL' in the stack and press enter, I get Error: Syntax error
I tried to type "config:equations.csv", load in the stack and then press RCL - I get recall menu with all my variables
I was able to type "config:equations.csv" to stack and then run RCL from the catalog menu. This is not very friendly, maybe I am missing something.

I had a couple of ideas, which are not essentials, but they can help usability. Thank you for this great development.

- How about, when we click on XEQ (thus evoking the ' ' ), a menu shows up with frequent used characters for this type of entry? Specially useful if we dont have the overlay.

- When clicking Shift Prog and then STO, we get the store command, but when click Shift Prog then RCL it pulls the recall menu. It would be useful to have consistent.

- Is it possible to create an equation and STO it directly into the equantions.csv without accessing the csv file? the stack holds the equation in natural language but the config file uses internal language (similar, but not as good for usability) a hotkey in the Solver menu to add or edit equations would be very useful

- How do you feel about the implementation of navigation like menu found in HP48G and 50G ?

c3d, 2024-06-25:
Hi @guirossibrum, thanks ffor your interest in DB48X.

> I tried to type the following '"config:equations.csv"RCL' in the stack and
> press enter, I get Error: Syntax error

This is not correct RPL syntax because of the quotes. You should try without the quotes, i.e. `"config:equations.csv" RCL`

> I tried to type "config:equations.csv", load in the stack and then press RCL -
> I get recall menu with all my variables I was able to type
> "config:equations.csv" to stack and then run RCL from the catalog menu. This
> is not very friendly, maybe I am missing something.

I suspect you have no overlay. The RCL key on the DM42 is labelled VAR on DB48X, and shows the variables menu, as you noticed.

The `RCL` command can be typed alphabetically, or is found in the `MEM` menu (shift-shift `STO`), with the label `Recall`.

c3d, 2024-06-25:
> How about, when we click on XEQ (thus evoking the ' ' ), a menu shows up with
> frequent used characters for this type of entry? Specially useful if we dont
> have the overlay.

The "frequently used stuff" key is the `ToolsMenu` on the `A` key, which is labelled `Σ+` on the DM42. When you are in expression mode, it gives you the `RealMenu`.

The frequently used characters for various use cases can be found in the CHAR menu (shift-shift-2).

Do you have another suggestion for what should be shown in RPL expressions?

c3d, 2024-06-25:
> When clicking Shift Prog and then STO, we get the store command, but when
> click Shift Prog then RCL it pulls the recall menu. It would be useful to have
> consistent.

That's because RPL does not use `RCL` nearly as often as RPN on the HP42, so it's not a primary key. The key labelled `RCL` on the DM42 is labelled `VAR` on DB48X and invokes the `VariablesMenu`

That being said, if you are in program mode, and have the `VariablesMenu` shown, then if the variable associated with F1 is `ABC`, then

- F1 will enters `ABC` on the command line, evaluating `ABC`
- Shift-F1 enters `'ABC' Recall` on the command line, recalling `ABC`
- Shift-Shift-F1 enters `'ABC' Store` on the command line, storing into `ABC`.

c3d, 2024-06-25:
> Is it possible to create an equation and STO it directly into the
> equantions.csv without accessing the csv file? the stack holds the equation in
> natural language but the config file uses internal [...]

The `equations.csv` is intended to store "standard" equations. Normal equations can be stored in the `VAR` menu just like any other object. For example, you can store `1+sin X` as an expresssion in a variable called `SineWave`.

The standard equations can be edited using the `Store` and `Recall` method on
the `config:equations.csv` file from the calculator or from a PC. The CSV file
format was chosen precisely so that it would be relatively easy to enter a large
set of standard equations from a PC, and then easily solve them on the
calculator. The constraints on CSV file parsing required me to store equations
as CSV text fields.

That being said, adding commands to store or recall standard equations is a good idea. Future feature.

c3d, 2024-06-25:
> How do you feel about the implementation of navigation like menu found in
> HP48G and 50G ?

I personally don't like it, because it's neither efficient nor scalable.
For efficiency, direct menus are faster.
For scalability, the catalog with autocompletion scales better.

And the on-line help provides the additional documentation for commands and topics.

guirossibrum, 2024-06-25:
Thank you. I love the work you are doing. It is a great effort to keep the
48 legacy alive and bring to the next level for future generations

Thanks

Guilherme Rossi Brum

closed 2024-06-27 by c3d