c3d/db48x #999: Compiling on Linux

JohnRobson, 2024-06-25:
I appreciate any support. I'm trying to compile the simulator on Arch Linux.

I installed: arm-linux-gnueabihf-gcc arm-linux-gnueabi-gcc

I did: git clone --recursive git@github.com:c3d/db48x.git

But I'm getting this error:

`
$ make sim
mkdir -p build/dm42/release
cd recorder && make TARGET=
make[1]: Entering directory '/tmp/db48x/recorder'
touch build/dm42/release/.exists
cd tools/ttf2font; make TARGET=
cd tools/decimize/; make
make[1]: Entering directory '/tmp/db48x/tools/ttf2font'
make[1]: Entering directory '/tmp/db48x/tools/decimize'
echo "#define DB48X_VERSION \"0.7.8\"" > build/dm42/release/version-0.7.8.h
cp build/dm42/release/version-0.7.8.h src/dmcp/version.h
cd sim; qmake db48x.pro -o db48x.mak CONFIG+=release 
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/tmp/db48x/recorder'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/tmp/db48x/tools/ttf2font'
tools/ttf2font/ttf2font -s 48 -S 80 -y -10 EditorFont fonts/FogSans-ddd.ttf fonts/EditorFont.cc
make: tools/ttf2font/ttf2font: No such file or directory
make: *** [Makefile:150: fonts/EditorFont.cc] Error 127
make: *** Waiting for unfinished jobs....
Info: creating stash file /tmp/db48x/sim/.qmake.stash
make[2]: Entering directory '/tmp/db48x/tools/decimize'
[BEGIN]          opt linux in [top]                                            
[COMPILE  1/1]   decimize.cpp                                                  
g++: error: unrecognized command-line option ‘-mthumb’
g++: error: unrecognized command-line option ‘-mfloat-abi=hard’
g++: error: unrecognized command-line option ‘-mfpu=fpv4-sp-d16’
make[2]: *** [../../recorder/make-it-quick/rules.mk:457: /tmp/db48x/tools/decimize/.build/linux/opt/decimize.cpp.o] Error 1
make[2]: Leaving directory '/tmp/db48x/tools/decimize'
`

Thank you so much.

JohnRobson, 2024-06-26:
I found that "those particular options are only available on the 32bit compiler, and this is the aarch64 gcc, so they do not apply. So if you are trying to port an "old" project to 64bit you'd want to remove them from your make environment."

so I replaced: #CPUFLAGS += -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16
to: CPUFLAGS += -march=armv7e-m

but still with several errors, like:

/usr/bin/ld: cannot open linker script file src/dm42/stm32_program.ld: No such file or directory

But the file is there: $ ls -lha src/dm42/stm32_program.ld
-rw-r--r-- 1 us us 4.8K Jun 25 15:24 src/dm42/stm32_program.ld

I think the author should compile it on Linux at least one time and fix these issues, I'm interested in building the simulator to run on Linux and share it.

prismv, 2024-06-26:
With a fresh checkout I can confirm that simply `make sim` does not work (anymore). There are several files in .gitignore that were once compiled successfully, but `make clean` does not delete them to get a completely virgin compilation:

- tools/ttf2font/ttf2font
- tools/decimize/decimize
- sim/gcc111libbid.a

If I start with fresh clone (here using depth to get only a shallow copy) without full git credentials there might be also problems with submodule pulling (so simple https access does not work for all submodules ...).

```
ssh-agent bash -c 'ssh-add ~/.ssh/whatever_id; git clone --recursive --depth 1 git@github.com:c3d/db48x.git'
cd db48x
for d in {recorder,tools/ttf2font,tools/decimize}; do pushd $d && make && popd; done
```

Now `make sim` should work, but there seems to be no proper rule to generate `sim/gcc111libbid.a`. @c3d I have a several months old version of this file in my older clone, but I don't know anymore how to generate this. @JohnRobson Sorry, at this point I have to give up due to other tasks.

dm42/stm32... files belong to the build of the firmware for the DM42 calculator. If I am not mistaken this should not belong to simulator code.

JohnRobson, 2024-06-26:
Thank you so much @prismv .

Yes, still with errors:  https://paste.debian.net/1321598/

`recorder.c:1997:20: error: conflicting types for ‘sigaction’; have ‘void(int,  void (**)(int), void (**)(int))’`

I hope @c3d  helps to fix these compiling issues.

c3d, 2024-06-27:
> But I'm getting this error: https://paste.debian.net/1321504

Indeed, it looks like there is a problem rebuilding `ttf2font`.

For now, the workaround is to first build it explicitly:

```
cd tools/ttf2font
make
cd ../..
make
make sim
```

That sequence worked for me. Of course, it should not be necessary, and I will fix it.

c3d, 2024-06-27:
> I found that "those particular options are only available on the 32bit
> compiler, and this is the aarch64 gcc, so they do not apply. So if you are
> trying to port an "old" project to 64bit you'd want to remove them from your
> make environment."

> so I replaced: #CPUFLAGS += -mthumb -march=armv7e-m -mfloat-abi=hard
> -mfpu=fpv4-sp-d16 to: CPUFLAGS += -march=armv7e-m

No, this is wrong. The options in question are for the _cross compiler_, the `arm-none-eabi-gcc` that is used to build a _32-bit_ executable for the DM42 or DM32. So you should leave them as is.

The simulator uses other options, which are defined by the `sim/db48x.pro` file, and is built using whatever Qt picks up for your particular distro, either gcc or clang.

> but still with several errors, like: /usr/bin/ld: cannot open linker script
> file src/dm42/stm32_program.ld: No such file or directory

If you see that, then you are not building the simulator, you are (incorrectly) building the binary for the DM42. That's the context where you need tht script.

> But the file is there: $ ls -lha src/dm42/stm32_program.ld -rw-r--r-- 1 4.8K Jun
> 25 15:24 src/dm42/stm32_program.ld

> I think the author should compile it on Linux at least one time and fix these
> issues, I'm interested in building the simulator to run on Linux and share it.

I think you presume incorrectly that the author does not build on Linux "at least one time" ;-) See for example https://github.com/c3d/db48x/issues/853.

The problem is that I do not build from scratch. I had to do a `git clean -dfx` to reproduce your problem in my build environment.

c3d, 2024-06-27:
I confirmed that these build dependency problems also exist with a clean build on macOS. I'm investigating, since normally the dependencies are there in the makefile so that it should work, and I'm pretty sure it used to work.

c3d, 2024-06-27:
This should be fixed on `dev` now. There was an easy problem and a subtle one.

The easy problem was that there was still a reference to the Intel decimal library in the `db48x.pro` file. Unless you start with a clean directory or `git clean -dfx`, that library file was never removed, notably by `git clean`, so I never noticed that I was depending on it.

The more subtle problem is a bad interaction between the environment variables that the outer `Makefile` sets and the expectations of the inner `make-it-quick` makefiles used for the tools. The invokation of the sub-makefiles with `TARGET=` on the command-line was supposed to workaround that, but the correct workaround is to pass `TARGET=opt` to force an optimized build of the tool.

JohnRobson, 2024-06-27:
> Indeed, it looks like there is a problem rebuilding `ttf2font`.
> 
> For now, the workaround is to first build it explicitly:
> 
> ```
> cd tools/ttf2font
> make
> cd ../..
> cd recorder
> make clean
> make
> cd ..
> make
> make sim
> ```

Thanks, is improving, but still with error:

`recorder.c:1997:20: error: conflicting types for ‘sigaction’; have ‘void(int,  void (**)(int), void (**)(int))’`

https://paste.debian.net/1321706/

closed 2024-06-30 by c3d

JohnRobson, 2024-06-30:
I don't understand why this issue is being closed if the compilation on Linux is still not working. Is there a genuine desire to solve issues or just to close them?

JohnRobson, 2024-06-30:
still not working:

`recorder.c:1997:20: error: conflicting types for ‘sigaction’; have ‘void(int,  void (**)(int), void (**)(int))’`

commit 87c60c72 (2024-06-27): build: Fix the clean build

Fix clean build of simulator with `make sim`:

- Pass `TARGET=opt` instead of `TARGET=` when building tools
  to ensure the tool gets properly built irrespective of environment

- Remove reference to obsolete Intel decimal library, which was still
  hanging around in my build directories.

Fixes: #999
commit 0dd5a753 (2024-06-27): build: Fix the clean build

Fix clean build of simulator with `make sim`:

- Pass `TARGET=opt` instead of `TARGET=` when building tools
  to ensure the tool gets properly built irrespective of environment

- Remove reference to obsolete Intel decimal library, which was still
  hanging around in my build directories.

Fixes: #999
commit 35e189dc (2024-06-30): build: Fix the clean build

Fix clean build of simulator with `make sim`:

- Pass `TARGET=opt` instead of `TARGET=` when building tools
  to ensure the tool gets properly built irrespective of environment

- Remove reference to obsolete Intel decimal library, which was still
  hanging around in my build directories.

Fixes: #999