c3d/db48x #1001: Key repeat sometimes gets stuck with the solver menu open

mahi2003, 2024-06-28:
Sometimes, after starting the solver, tapping a number button to enter a value does not seem to register the release event and keeps repeating the number (as if you kept the button pressed).

I'm not sure yet what triggers this because I cannot reproduce it with every equation, nor does it happen on USB power, but I can do it quite consistently as follows (even after a reset):

* Make sure to run on battery power (DM42)
* Start the solver for the IdealGas equation from the equations menu (`MAIN` `Eqns` `Phys`)
* Tap any number button (just a short press). The calculator enters edit mode, prints the entered digit and then (already after the button was released) repeats the number indefinitely
* Press any other number button → The issue stops and normal entry is possible
* Press `EXIT` until the solver menu disappears
* Tap a number button again → The issue does not occur
* `LastMenu` (to reopen the solver menu)
* Tap a number button again → The issue returns

It happens only when transitioning to entry mode with the solver menu open, but not always. Pressing for example `XEQ` does not trigger it. Nor does `"TEXT"`, but enabling alpha mode and then pressing any button (with alpha) will trigger it.

Pressing the buttons just a tad longer also prevents the issue from happening. Judging by that and the fact that it does not happen on USB power, it is most likely a timing issue. I guess it won't happen in the simulator.

c3d, 2024-06-30:
I have been unable to reproduce.

However, I made a number of display refresh fixes related to the problem with the annunciators (shift and alpha) and the management of the timers. I suspect one of them may have fixed that specific issue.

Let's see if you can reproduce with the release I do today, and if you can't, we'll close this issue.

mahi2003, 2024-07-01:
✔️ I've not been able to reproduce it on 0.7.9. You may close this issue.