c3d/db48x #1006: 0.7.9 installation fail

guirossibrum, 2024-07-01:
Hello Christophe. I get error: bad argument type 
E. 54937758936840882/5hJ on first level of stack. 
it is very hard to come out of the error into system menu but I was able to and reset DM42 to original state. After reinstating v0.7.9 I get again the same error

guirossibrum, 2024-07-01:
BTW, happens immediately at first boot

guirossibrum, 2024-07-01:
[image]

guirossibrum, 2024-07-01:
One additional comment - I was able to locate the problem. The state file was loaded with the update and it was probably incompatible:

8.98755178736817640E17_J
'E' STO

10_kg
'm' STO

5
'T' STO

1.22500E2
'H' STO

ⒺPhys
'Equation' STO

:E:44937758936840882/5_hJ
ImproperFractions
ShowBuiltinUnits
SquareMenus
TextResultDisplay
5 Fix
5 DisplayDigits
Bin
4 ResultFont
4 StackFont
4 EditorFont
8 WordSize
0 ErrorBeepFrequency
SolvingMenu
 HomeDirectory

c3d, 2024-07-09:
I will reopen the issue because the state file causes a crash in the simulator. So there is still something wrong with it.

c3d, 2024-07-09:
The part that is curious in the state file is this:

```
ⒺPhys 'Equation' STO
```

The `Phys` equation does not exist, but there is an equation menu by that name, so the lookup finds a match.

An equation menu does not have a definition, which causes an error if you read from a config file, or a crash if you read it from memory.

There is probably a state save problem as well with an index mismatch that causes it to save the equation with the wrong name.

commit 3157b673 (2024-07-12): constants: Skip menu entries

The menu entries do not have a definition.
This causes a crash on simulator when there is no config file,
and some kind of weird error on the real machine.

Fixes: #1006
closed 2024-07-17 by c3d