c3d/db48x #1008: CLR menu does not open while in program mode

guirossibrum, 2024-07-06:
In program mode you can select a menu and enter the command as part of a program. This is not possible with the CLR menu.

mahi2003, 2024-07-07:
I think the problem here is that the `←` key has function DEL(ete) on the first shift while in entry mode. This overlaps with the CLR menu, but I'd hate to lose the DEL function.

* ← (backspace): Delete character in front of cursor
* DEL: Delete character after cursor

guirossibrum, 2024-07-07:
The Del button is the shifted alpha, so you would not lose it.
The regular shift should open the menu to keep consistency With the
remaining of the keyboard.

Guilherme Rossi Brum

c3d, 2024-07-09:
> I think the problem here is that the `←` key has function DEL(ete) on the
> first shift while in entry mode. This overlaps with the CLR menu, but I'd hate
> to lose the DEL function.
>
> * ← (backspace): Delete character in front of cursor
> * DEL: Delete character after cursor

That is indeed the reason.

> The Del button is the shifted alpha, so you would not lose it.  The regular
> shift should open the menu to keep consistency With the remaining of the
> keyboard.

The alpha mode is not needed. The DEL function is available in normal mode with
a single shift.

The same problem also exists with the second shift, which in interactive mode goes one level up in the directory stack (UPDIR), but when editing enters a new line.

These behaviors are intentional, because the corresponding operations are quite useful in interactive mode, but not used that often in programs. The corresponding commands are available in the `StackMenu` (`ClearStack`), `MemoryMenu` menu (`Purge` and `UpDir`), the `PlotMenu` (`ClLCD`), `StatisticsMenu` (`ClearData`).

I think only `err0` is missing in menus, but then so are `IfErr`, `DoError`, `ErrorMessage` and `ErrorNumber`. I added the missing features to the `Debug` menu.