c3d/db48x #1011: Stack pick

guirossibrum, 2024-07-06:
On the HP48 when we click the stack menu we can move up the stack with the arrows and then pick the value to be copied to the first level of stack. On DB48X you put the stack number in the stack and the run PiCK(which you can’t do in the Hp48)

Is there a way to navigate up the stack to see the contents?this is specially used with infinite size stack. 
thank

mahi2003, 2024-07-07:
`<level>` `PICK` works on the HP48 too, but I agree that picking (and echoing) items from the stack with the arrow keys is a must have feature on an infinite stack. I also like that you can quickly make a list of 1-n stack items.

Judging by the "STACK" text next to the up/left arrow button on the DB48X overlay, I assume this is an upcoming feature.

guirossibrum, 2024-07-07:
I can’t make the <level> PICK work on the HP48 though. When I enter the
stack menu, I get the arrow on the left side of the screen and the arrow
overrides any content on the first level of the stack.
Do you mean in program mode?

Guilherme Rossi Brum

mahi2003, 2024-07-07:
Maybe I misunderstand you, but on the HP 48 (and 50) entering `5` `PICK` will pick (copy) the 5th stack level — exactly the same as in DB48X. If you don't want to type `PICK`, you can also find it in the STACK menu (HP 48: left shift, up arrow).

But most of the time I use the up arrow stack browser for the reason you state: with an infinite stack, the item you need may be out of view. So, being able to scroll through the stack and pick or echo items is a great feature that would be nice to have in DB48X.

guirossibrum, 2024-07-07:
No. You are right. I never tried that way before. Thanks

Guilherme Rossi Brum

c3d, 2024-07-09:
The feature is indeed coming. And since it's at least the third request, I might put it higher on my priority list.

commit 85d8599c (2024-07-12): stack: Interactive stack

Draw the interactive stack when using the 'up' arrow.

Fixes: #1011
commit b272158d (2024-07-12): stack: Interactive stack menu

Add menu functions for the interactive stack

Fixes: #1011
commit 45ef03c4 (2024-07-12): stack: Interactive stack menu

Add menu functions for the interactive stack.

Compared to the HP implementation, the interactive stack that was
implemented here adds stack sort and revert features.

Fixes: #1011
closed 2024-07-17 by c3d