c3d/db48x #1017: `_hms`, `_date` and `_d` do not work with values on the stack

mahi2003, 2024-07-10:
When working with units, you can append a unit to an existing value in stack level 1. For example: `5` `ENTER` `m` → `5_m`

This does not seem to work for `_hms` (time), `_date` (date) and `_d` (days).

`5` `ENTER` `_d` → :x: Syntax error

You can only use it in entry mode

`5` `_d` `ENTER` → `5_d`

c3d, 2024-07-10:
I do not fully understand the description. When you write `5 ENTER m`, what is the key sequence corresponding to `m`? Is it the first function key in the `Length` unit menu? If that is the case, then it also works with the `d` entry in the `Time` menu.

I suspect that the `_dms`, `_date` and `_d` are the corresponding entries in the `TimeMenu` and `DateMenu`. Indeed, those are only editor entry shortcuts, they are not operations.

There are operations that do the same thing. In the `TimeMenu`, for example, F1 is the data entry `_hms`, and the conversion to HMS is on key F3 (`→HMS`). Similarly for time with `→Date` and `→Time`.

I see this more as a documentation problem and maybe a usability issue than as a bug, in the sense that it's working as intended. However, please feel free to suggest a way this could work. Maybe the `→Date` menu could directly enter `_date` in data entry mode?

mahi2003, 2024-07-11:
You are right. I misunderstood the purpose of `_hms`, `_date` and `_d` in the `TimeMenu` and `DateMenu`.

I now realize I can enter a time with `→HMS`, but what's the equivalent for a date? Say I want to make `20240711` a date. There isn't a `→YMD` command and `→Date` just updates the calculator date.

c3d, 2024-07-13:
You are correct that there is no equivalent function to `→HMS`, mostly because it does not exist in the HP48 so I did not think about making one. I will add that in an upcoming release. I will also see if I can have the `_hms` and similar menu entries act as a conversion function when not editing. I need to think about the right way to do that.

For now, you can use the `Date+` menu entry with a number of days of 0, i.e. `19681205 0 Date+` will give you a date. But as you can infer, it is quite easy to add that feature as a standalone function.