c3d/db48x #1022: Interactive stack: Block keyboard input?

mahi2003, 2024-07-15:
I noticed that while in the interactive stack the calculator still accepts normal keyboard entry. For example, with stack...

```
1
2
3
```

... press the left/up arrow to enter the interactive stack and select, for example, the second stack level.

* Now press the `SIN` button and the sinus of stack level 1 is calculated
* Press the `x↔y` button and stack levels 1 and 2 are swapped
* Press the `+` operator and stack levels 1 and 2 are added
* Press numeric button `1` and the calculator enters entry mode even though still in the interactive stack

On the HP 48/50 the interactive stack blocks all keyboard entry other than those related directly to the stack manipulation (the soft keys, up, down, backspace, ENTER, ON, NXT, PREV, COPY).

c3d, 2024-07-19:
You are correct, this is the expected behavior. The key blocking needs to be implemented.

closed 2024-07-22 by c3d

mahi2003, 2024-07-23:
✔️ Confirmed fixed in 0.7.11

commit 9a0655a6 (2024-07-19): ui: Block user input while using interactive stack

While using the interactive stack, we don't want a key like `sin` to
activate the corresponding function. We also don't need to activate
built-in help.

Fixes: #1022