c3d/db48x #1023: Interactive stack suggestion: Access from entry mode to echo items from the stack into the current entry

mahi2003, 2024-07-15:
Sometimes it can be useful to copy a value or object from the stack to the current entry. Say for example you are writing a program and want to include an equation you wrote earlier. It would be nice if it was possible to access the interactive stack from within entry mode to echo/copy an item from the stack to the current cursor position.

The left/up arrow cannot be used for this, as it is already in use for cursor movement in entry mode. Maybe, an entry for the interactive stack can be added to the `StackMenu`?

The HP 48 had a similar feature (albeit a little bit broken because it did work when the current entry had multiple lines). For some reason it was not carried over to the HP 50.

c3d, 2024-07-19:
Thanks for the report. I also had [a comment][edit] that the `Edit` feature is broken in that it does not let you edit a higher level in the stack directly.

I will rework this for the next release to include this feedback. I agree with you that this could be useful. That being said, it's possible to achieve the same effect (slightly more slowly) by using the copy/paste feature. It is possible that this is the reason that feature disappeared on the HP50, since that calculator also had true copy/cut/paste, and people are more familiar with this paradigm.

On the HP50, you would use the interactive stack to bring the object you want into the editor, then use BEGIN/END to select a section, then COPY, then exit, then edit the object and use PASTE to insert the (part of the) object you want.

On DB48X, you can do something like that, with the added benefit that you have 8 levels of command-line history (a part that I intend to rework as well).

[edit]: https://www.hpmuseum.org/forum/thread-20157-post-189389.html#pid189389

c3d, 2024-07-22:
This was added as part of the rework of the interactive stack menu and `EditMenu`.
There is now a `Stack` menu entry while editing.

mahi2003, 2024-07-23:
Thank you!

There's one 'minor' issue that prevents this from being really useful: Each time I `Echo` or `EchoNSp` an item from the stack it is appended at the end of the current entry. Not where the cursor is.