c3d/db48x #1024: Interactive stack: Menu items are painted incorrectly

mahi2003, 2024-07-15:
* Have at least one item on the stack
* Open the `SolverMenu`. Note how the the `Diff`, `Poly`, `Solve`, `Num` and `Symb` menu entries have a folded corner to indicate they are folders (or sub menus)
* Press left/up arrow to enter the interactive stack. The menu changes to that of the interactive stack. The menu entries that are in the same location as the previous sub menus (`Keep`, `Roll↑`, `Roll↓`, `Pick`and `→List`) now have the same folded corners

Same issue, but differently:

* Have at least one item on the stack
* Press left/up arrow to enter the interactive stack. The menu changes to that of the interactive stack
* Press the `VAR` button on the keyboard and observe what happens to the shifted menu entries: `DupN`, `DropN`, `Keep`... become `DupN►`, `DropN►`, `Keep►`

c3d, 2024-07-19:
Good catch. If my theory is correct, I think that the same thing should be true with the Help menu.

closed 2024-07-22 by c3d

mahi2003, 2024-07-23:
✔️ Confirmed fixed in 0.7.11

commit 4876991a (2024-07-19): ui: Do not draw menu markers when displaying interactive stack

The interactive stack has a special menu, and like the help menu, it
should not show the menu markers.

Fixes: #1024