c3d/db48x #1040: equations: Add equations from the HP50 equation library

c3d, 2024-07-22:
The equation library in the HP50 Advanced Reference Manual is a good starting point for interesting equations and their graphical representation.

Wiljea, 2024-08-12:
I would like to help here to enter and document HP50G (since I have some free time now) but I don't how besides the following:
create a tagged list of eqns for instance:
**Ohm's Law & Power:{'V=I*R' 'P=V*I' 'P=I^2*R' 'P=V^2/R'}**
But I can't see where to save them on the calculator. Being on the Win10 platform, I can't edit correctly the csv file in (undefined spurious characters see below) in Notepad++ where surprisingly your recent addition (Columns & Beams eqns) doesn't appear. 

Also is it necessary to attach units explicitly as you did for in the case in your CSV file:
**IdealGas,                     "'(P_Pa)*(V_m^3)=(n_mol)*â’¸R*(T_K)'"**

Wiljea, 2024-08-13:
ATTEMPT#1
I tried to edit the file equations.csv in my directory \\wsl.localhost\fedoraremix\home\Wiljea\db48x\config
to add the following:
"Coulomb'sLaw",                 "'(F_N)=1/(4*Ⓒπ*Ⓒε0)*((q1_C)*(q2_C)/(r_m)^2)'" 
but that doesn't appear in the emulator (where I copied the "Ⓒ"R constant prefix character from the IdealGas law of the same file). Maybe the emulator reads this file only at start-up (during make sim)? What to do then ?

Wiljea, 2024-08-15:
New eqns completed for Fluids, Forces & Energy, Gases and Heat Transfer. I corrected few errors in Electricity. It is tedious work to add and check the physical units. I am transmitting a newer version (Equations_V2.csv) to replace the former one at the address christophe@dinechin.org (is this address active ?). I am expecting your feedback to see if this works.

c3d, 2024-08-16:
Hi @Wiljea, thanks for offering to help.

Constants, equations and library (collectively "library items") follow the same logic: there is a built-in set, defined in the source code, and you can customize it with the matching `.csv` file in the `config` directory. The two are combined to build what you see in the menus, unless you set the corresponding setting to disable the use of built-in constants/equations/library items when the file is found (this is in case you want to completely override the built-in ones).

The `.csv` file is read relative to the current directory. If you launch the simulator from the `sim` directory and not the `db48x` directory, then the `config` subdirectory will not be found. You can either change directory, create a symlink to the `config` directory, or copy it under `sim`. That should solve your problem loading the file. If not, you should have a `file_error` message somewhere on the console telling you which file was not read and why (e.g. "file not found" or something else).

I have received your email and I will gladly accept the help. That being said, I am currently on vacation in a spot that has very poor 3G connectivity, so I will most likely not really act on it before Sept 1st. Watch the next few releases :-)

If you want to do it "the right way" and are ready to go into DB48X programming, you can edit the file "equations.cc", adding the data you generated for "equations.csv" in C format. That would add it to the built-in equations. I would appreciate if you could also add the corresponding tests in the `tests.cc` file, ideally taking values and solutions from the HP50G advanced reference manual (it's available as a PDF on the HP Museum web site). This will ensure that things work as intended.

To run the test suite for a specific topic, use `sim/db48x.exe -T<section>` where `<section>` is the section testing what you are interested in. For example, "Columns and Beams" is a section called `colnbeams`.

c3d, 2024-08-16:
BTW, you mentioned that you were running on Win10, which is a platform I do not build for personally. How do you run the simulator there? Are you using the Linux subsystem for Windows, or buildiing the Qt version directly on Win10? If the latter, would you mind adding a Wiki article explaining how you did it, because I know that others asked about it.

Wiljea, 2024-08-16:
Thanks for your reply Christophe,
I apologize for disturbing you during your well-deserved vacation. I will try to be brief and concise so that I can then move forward on my own.
  1. Unfortunately, I don't build the QT version on Win10, because I actually use a Linux console by the Fedore Remix for WSL environment (as pointed out by Rowdy from the SwissMicros forum) and I almost always manage to build the db48x emulator correctly (I sometimes have to do it several times to create the latest version of the F/W).
  2. Unfortunately also I know very little of the C language (being a physicist, I am more interested in numerical methods, I especially know Fortran, Focal-Free42 (DM42, HP41, etc.) and more recently RPL which I learn to appreciate much for its power).
  3. BUT, if I understood correctly, it is directly in the equations.cc file that I need to intervene. More precisely, between lines 53 and 75 of this version. So what I am about to do is to reproduce the unit block going from lines 54 to 65 inclusively for each of the equation sections to import. So I continue what I started by attaching the units of each variable and inserting one equation per line in the equations.cc file.
  4. If this is indeed the case, since I found said file I should be able to do it (before your return) without too much problem by completely transferring the HP50g library and supplementing it with a few additional equations from my believed.
  5. I don't know yet what you mean by adding the corresponding tests in the tests.cc file (I didn't look as of now), but once my first task completed, I will study the situation together with the HP50g reference manual to see if a able to make sense about that.

Thanks for your attention and I hope you enjoy the rest of your vacation.

Wiljea, 2024-08-16:
Thanks for your reply Christophe,
I apologize for disturbing you during your well-deserved vacation. I will
try to be brief and concise so that I can then move forward on my own.
  1. Unfortunately, I don't build the QT version on Win10, because I
actually use a Linux console by the Fedora Remix for WSL environment (as
pointed out by Rowdy from the SwissMicros forum) and I almost always manage
to build the db48x emulator correctly (I sometimes have to do it several
times to create the latest version of the F/W).
  2. Unfortunately also I know very little of the C language (being a
physicist, I am more interested in numerical methods, I especially know
Fortran, Focal-Free42 (DM42, HP41, etc.) and more recently RPL which I
learn to appreciate much for its power).
  3. BUT, if I understood correctly, it is directly in the equations.cc
file that I need to intervene. More precisely, between lines 53 and 75 of
the present version. So what I am about to do is then to reproduce the unit
block going from lines 54 to 65 inclusively for each of the equation
sections to import. So I continue what I started by attaching the units of
each variable and inserting one equation per line in the equations.cc file.
  4. If this is indeed the case, since I found said file I should be able
to do it (before your return) without too much problem by completely
transferring the HP50g library and supplementing it with a few additional
equations from my belief.
  5. I don't know yet what you mean by adding the corresponding tests in
the tests.cc file (I didn't look as of now), but once my first task is
completed, I will study the situation together with the HP50g reference
manual to see if I am able to make sense about that.

Thanks for your attention and I hope you enjoy the rest of your vacation.

Jean Wilson
CC Github

Wiljea, 2024-08-21:
PROGRESS REPORT:
Having added several new equations to the existing sections and also a brand new section on relativity, I already exceed a total of 400 equations. This represents a 27% increase over the HP50g library. I work in 2 iterations to eliminate typos and to ensure consistency of parentheses (opening & closing) and the logical coherence of physical units. 
The second iteration ends by incorporating them into the C code of the equations.cc file. The task is 30% complete as of 24-08-21.

Wiljea, 2024-08-27:
PROGRESS REPORT (24-08-26):
My last version contains 506 equations which corresponds to an increase of 60% compared to the initial library (315). I took care to add the units as they appear in the HP50G examples. Essentially I completed the existing chapters by adding several equations that I believed were relevant to deepen the topics with a brand new chapter of 15 sections on relativity with 86 equations. 

To cover the pre-university level in science, a chapter on modern physics is missing which should include at least twenty equations.