c3d/db48x #1046: Feature Request - can the menus buttons get a toggle to allow them to be persistent?

spiff72, 2024-07-24:
I like to see the menu buttons all the time - specifically to be able to see my variables.

Currently, if you press the exit key twice, the menu disappears, and I have to press the VAR button again to bring it back.  I also find it jarring when I have a folder that contains no variables - I would rather see blank buttons across the bottom than have it disappear when I have the variables displayed and switch to a different folder that is empty.

So my ask is to see if it's possible to have a toggle in settings that allows for persistence of the variables menu buttons.  I'm not sure if this is valid outside of the variables menu, but thought I would ask!

c3d, 2024-07-25:
I see two slightly different requests here:

- A preference controlling whether `EXIT` clears menus
- A preference controlling whether blanks menus should show on screen.

I'd rather make it two distinct preferences.

c3d, 2024-07-25:
Implemented with two pairs of settings:

- `HideEmptyMenu` / `ShowEmptyMenu`
- `ExitClearsMenu` / `ExitDoesNotClearMenu`

spiff72, 2024-07-25:
Fantastic - thanks!

commit 2263a012 (2024-07-27): ui: Add settings to hide/show empty menus and clear menu on EXIT

By default, `EXIT` clears the current menu when not editing.
The `ExitDoesNotClearMenu` setting blocks this behaviour.

By default, empty menus are not shown, leaving more room for the
stack. The `ShowEmtyMenus` ensures that empty menus entries are shown
on the screen.

Fixes: #1046

Suggested-by: spiff72 <mgobluevictor@gmail.com>
closed 2024-07-29 by c3d