c3d/db48x #1064: Unit conversions with additive factor

c3d, 2024-07-28:
We need a way to implement conversions with an additive factors, e.g. Kelvin to Celsius and conversely.

Wiljea, 2024-08-09:
Suggestion:
The idea is to relate in each case (by providing the corresponding equations in the csv file) to or from the most fundamental or primary SI unit of the same category.
For instance, in the case of **temperature** we have to relate to **kelvin (K)** and use it in intermediate step for conversion between **Tˍ°C** and **Tˍ°F** the relevant equations are:
            **Tˍ°C=TˍK+273.15**  and  **Tˍ°F=9/5*(TˍK+273,15)+32**
and their converse:
            **TˍK=Tˍ°C-273.15**  and  **TˍK=5/9*(Tˍ°F-32)-273.15**

Wiljea, 2024-08-09:
This idea to include additional info (like equations and uncertainty) in the CSV files would also apply to the physical constants case in a future development. Indeed, the db48 platform (with its adjustable precision) provides the best imaginable tool to calculate physical quantities with the sufficient and maximum precision available. We have to remind that since 2019, the SI unit system was completely redefined and finally unified with this precise goal in mind. We have now precisely defined fundamental constants (like c, NA, h, etc.) and secondary ones which depend directly (through defining equations) on the former and on some precise measurement (of given constants and their actual uncertainty). For instance the magnetic permeability (**μo ± Δμo**) depends strictly on the measured value the affine structure constant (α±Δα). The relevant equations are 👍(thanks !) :
         **μo=2*α*h/(e^2*c)**   and   **Δμo/μo=Δα/α**
as simple as that !
Therefore, from the point view of a modern physicist, a universal calculator should provides for these complete information : Values, units, defining eqs for val and Δ.
There is no such hand-calculator to date (besides Maple or Mathematica which are heavy duty tools).
And we have this unique opportunity to create one. I might able to provide the relevant equations if you are interested. And you will understand my personal interests for random numbers, uncertainty evaluation, error propagation (under given prob. distr.) and interval arithmetic. (I am the same person as jeanwilson from the SwissMacros and HP Forum). These ideas and projects are for a future that I like very much to imagine !

c3d, 2024-11-02:
Adding the possibility to add an expression involving the source unit in the definition. For example, the definition of `°C` becomes `'(°C+273.15)'_K`. The unit conversion system detects this case and either uses the unit forward (in this conversion case) or backwards by using the `Isolate` feature (i.e. to turn `K=°C+273.15` into `°C=K-273.15`).

This more sophisticated approach is slightly more expensive to compute in the backrwards case, but it much more general, since it enables non-linear scales, like the [Dalton temperature scale](https://physics.stackexchange.com/questions/459851/john-daltons-temperature-scale#459863) or the [dB to power](https://en.wikipedia.org/wiki/Decibel) conversions.

commit 269fdc2b (2024-11-02): units: Implement non-proportional unit conversions

Add the possibility to add an expression involving the source unit in
the definition.

For example, the definition of `°C` becomes `'(°C+273.15)'_K`.

The unit conversion system detects this case and either uses the unit
forward (in this conversion case) or backwards by using the `Isolate`
feature (i.e. to turn `K=°C+273.15` into `°C=K-273.15`).

This more sophisticated approach is slightly more expensive to compute
in the backrwards case, but it much more general, since it enables
non-linear scales, like the [Dalton temperature scale][D] or  [... cut]