c3d/db48x #1066: flash overflowed

prismv, 2024-07-29:
No problems before on Archlinux due to program size, but when trying to make actual release 0.7.12 directly from git, this happens:

> /usr/lib/gcc/arm-none-eabi/14.1.0/../../../../arm-none-eabi/bin/ld: build/dm42/release/db48x.elf section `.data' will not fit in region `FLASH'

c3d, 2024-07-29:
Thanks for reporting. What is the version of your toolchain?

(It looks like your compiler is not packing things dense enough to fit)

prismv, 2024-07-29:
gcc 14.1.0 (it's already written in the path), so the newest released one.

My - not so smart - guess would be that on your machine some optimizations are turned on by default, but not explicitly enabled. Or some mysterious OS-specific variations of gcc?

prismv, 2024-07-29:
Will it help to share the output of `/usr/bin/arm-none-eabi-c++ -dumpspecs`?

prismv, 2024-07-30:
So far at least `make OPT=small` works (e.g. with dev 06f462d).

c3d, 2024-07-31:
I'm building with `10.3-2021.10`. It looks like there was some regression.

Could you share the output of the following command so that we get a better feel for how much we lost and which sections were impacted?

```
arm-none-eabi-objdump  -h build/dm42/release/db48x.elf

build/dm42/release/db48x.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .qspi         0003f500  90000000  90000000  000d0000  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  1 .rodata       000099a0  08050000  08050000  00010000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .text         0009a438  080599c0  080599c0  000199c0  2**6
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .ARM.extab    00000000  080f3df8  080f3df8  0010f500  2**0
                  CONTENTS
  4 .ARM          00000008  080f3df8  080f3df8  000b3df8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  5 .preinit_array 00000000  080f3e00  080f3e00  0010f500  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  6 .init_array   00000020  080f3e00  080f3e00  000b3e00  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  7 .fini_array   00000008  080f3e20  080f3e20  000b3e20  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  8 .data         00001098  10000000  080f3e28  000c0000  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  9 .bss          00000c74  10001098  10001098  000c1098  2**3
                  ALLOC
 10 .ARM.attributes 00000030  00000000  00000000  0010f500  2**0
                  CONTENTS, READONLY
 11 .comment      00000049  00000000  00000000  0010f530  2**0
                  CONTENTS, READONLY
 12 .debug_frame  00001f18  00000000  00000000  0010f57c  2**2
                  CONTENTS, READONLY, DEBUGGING, OCTETS

```

prismv, 2024-07-31:
After changing flash size to 705k in linker script to get the elf file this is the objdump:

```
arm-none-eabi-objdump  -h build/dm42/release/db48x.elf                                                                                          ~/repos/github/c3d--db48x(stable✗)@rapc2

build/dm42/release/db48x.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .qspi         0003f4e8  90000000  90000000  000b2000  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  1 .rodata       00009748  08050000  08050000  00001000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .text         000a5dc8  08059780  08059780  0000a780  2**6
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .ARM.extab    00000000  080ff548  080ff548  000f14e8  2**0
                  CONTENTS
  4 .ARM          00000008  080ff548  080ff548  000b0548  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  5 .preinit_array 00000000  080ff550  080ff550  000f14e8  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  6 .init_array   00000020  080ff550  080ff550  000b0550  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  7 .fini_array   00000008  080ff570  080ff570  000b0570  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  8 .data         00000d98  10000000  080ff578  000b1000  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  9 .bss          00000f40  10000d98  10000d98  000b1d98  2**3
                  ALLOC
 10 .ARM.attributes 00000030  00000000  00000000  000f14e8  2**0
                  CONTENTS, READONLY
 11 .comment      0000003c  00000000  00000000  000f1518  2**0
                  CONTENTS, READONLY
 12 .debug_line   00014f6a  00000000  00000000  000f1554  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 13 .debug_line_str 00000185  00000000  00000000  001064be  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 14 .debug_info   00024de5  00000000  00000000  00106643  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 15 .debug_abbrev 0000919b  00000000  00000000  0012b428  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 16 .debug_aranges 00000e08  00000000  00000000  001345c8  2**3
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 17 .debug_str    0000373a  00000000  00000000  001353d0  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 18 .debug_frame  00001c84  00000000  00000000  00138b0c  2**2
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 19 .debug_loclists 00015f71  00000000  00000000  0013a790  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 20 .debug_rnglists 0000108c  00000000  00000000  00150701  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
```

Besides the changed `stm32_program.ld` this is the current stable branch.

c3d, 2024-08-01:
The most relevant part is the size of the `.text` section.

In my build, it's `0009a438` (631864). In yours it's `000a5dc8` (679368). That's almost 50K extra.

c3d, 2024-08-01:
Just curious, are you building debug? There is a lot of additional debug information in there. That could explain the overflow.

prismv, 2024-08-02:
Nothing is changed compared to the repo. Following are the last few lines of `make` output of a fresh clone, i.e. `-DNDEBUG` in any g++ compilation step:

```
[...]
arm-none-eabi-g++ -c -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wno-packed-bitfield-compat -Wall -fdata-sections -ffunction-sections  -DNDEBUG -DDM42 -DHELPFILE_NAME=\"/help/db48x.md\" -D__weak="__attribute__((weak))" -D__packed="__attribute__((__packed__))" -Isrc/dm42 -Isrc/dmcp -Isrc -Idmcp/dmcp -O2 -Wno-misleading-indentation  -MD -MP -MF .dep/variables.o.d -fno-exceptions -fno-rtti -Wa,-a,-ad,-alms=build/dm42/release/variables.lst src/variables.cc -o build/dm42/release/variables.o
arm-none-eabi-gcc -x assembler-with-cpp -c -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wno-packed-bitfield-compat -Wall -fdata-sections -ffunction-sections  -DNDEBUG -DDM42 -DHELPFILE_NAME=\"/help/db48x.md\" -D__weak="__attribute__((weak))" -D__packed="__attribute__((__packed__))" -Isrc/dm42 -Isrc/dmcp -Isrc -Idmcp/dmcp -O2 -Wno-misleading-indentation  -MD -MP -MF .dep/startup_pgm.o.d dmcp/dmcp/startup_pgm.s -o build/dm42/release/startup_pgm.o
arm-none-eabi-gcc build/dm42/release/pgm_syscalls.o build/dm42/release/target.o build/dm42/release/sysmenu.o build/dm42/release/main.o build/dm42/release/EditorFont.o build/dm42/release/HelpFont.o build/dm42/release/ReducedFont.o build/dm42/release/StackFont.o build/dm42/release/algebraic.o build/dm42/release/arithmetic.o build/dm42/release/array.o build/dm42/release/bignum.o build/dm42/release/catalog.o build/dm42/release/characters.o build/dm42/release/command.o build/dm42/release/comment.o build/dm42/release/compare.o build/dm42/release/complex.o build/dm42/release/conditionals.o build/dm42/release/constants.o build/dm42/release/datetime.o build/dm42/release/decimal.o build/dm42/release/equations.o build/dm42/release/expression.o build/dm42/release/file.o build/dm42/release/files.o build/dm42/release/font.o build/dm42/release/fraction.o build/dm42/release/functions.o build/dm42/release/graphics.o build/dm42/release/grob.o build/dm42/release/hwfp.o build/dm42/release/integer.o build/dm42/release/integrate.o build/dm42/release/library.o build/dm42/release/list.o build/dm42/release/locals.o build/dm42/release/logical.o build/dm42/release/loops.o build/dm42/release/menu.o build/dm42/release/object.o build/dm42/release/plot.o build/dm42/release/polynomial.o build/dm42/release/program.o build/dm42/release/renderer.o build/dm42/release/runtime.o build/dm42/release/settings.o build/dm42/release/solve.o build/dm42/release/stack.o build/dm42/release/stats.o build/dm42/release/symbol.o build/dm42/release/tag.o build/dm42/release/text.o build/dm42/release/unit.o build/dm42/release/user_interface.o build/dm42/release/util.o build/dm42/release/variables.o build/dm42/release/startup_pgm.o  -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Tsrc/dm42/stm32_program.ld   -Wl,-Map=build/dm42/release/db48x.map,--cref -Wl,--gc-sections -Wl,--wrap=_malloc_r -o build/dm42/release/db48x.elf
/usr/lib/gcc/arm-none-eabi/14.1.0/../../../../arm-none-eabi/bin/ld: build/dm42/release/db48x.elf section `.data' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/14.1.0/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 784 bytes
/usr/lib/gcc/arm-none-eabi/14.1.0/../../../../arm-none-eabi/bin/ld: warning: build/dm42/release/db48x.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
make: *** [Makefile:446: build/dm42/release/db48x.elf] Fehler 1
```

prismv, 2024-10-31:
I had to use `make OPT=small` for several weeks, but now the ARCH rolling release changed something with the arm-none-eabi-* so that `make OPT=release` (or simply `make`) provides small enough programs without "flash overflow".

It seems to be a compiler version and/or distribution problem.

```
> make
[...]
arm-none-eabi-size build/dm42/release/db48x.elf
   text    data     bss     dec     hex filename
 878664    3672    3552  885888   d8480 build/dm42/release/db48x.elf
[...]
# Built 0.8.3Z
> arm-none-eabi-g++ --version
arm-none-eabi-g++ (Arch Repository) 14.1.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> arm-none-eabi-objdump  -h build/dm42/release/db48x.elf

build/dm42/release/db48x.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .qspi         00042150  90000000  90000000  00097000  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  1 .rodata       00010e00  08050000  08050000  00001000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .text         000838f0  08060e00  08060e00  00011e00  2**6
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .ARM.extab    00000000  080e46f0  080e46f0  000d9150  2**0
                  CONTENTS
  4 .ARM          00000008  080e46f0  080e46f0  000956f0  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  5 .preinit_array 00000000  080e46f8  080e46f8  000d9150  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  6 .init_array   00000020  080e46f8  080e46f8  000956f8  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  7 .fini_array   00000008  080e4718  080e4718  00095718  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  8 .data         00000e30  10000000  080e4720  00096000  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  9 .bss          00000de0  10000e30  10000e30  00096e30  2**3
                  ALLOC
 10 .ARM.attributes 00000030  00000000  00000000  000d9150  2**0
                  CONTENTS, READONLY
 11 .comment      0000001e  00000000  00000000  000d9180  2**0
                  CONTENTS, READONLY
 12 .debug_line   00016ccb  00000000  00000000  000d919e  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 13 .debug_line_str 00000185  00000000  00000000  000efe69  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 14 .debug_info   00025cd6  00000000  00000000  000effee  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 15 .debug_abbrev 000093bd  00000000  00000000  00115cc4  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 16 .debug_aranges 00000e10  00000000  00000000  0011f088  2**3
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 17 .debug_str    0000368b  00000000  00000000  0011fe98  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 18 .debug_frame  00001c28  00000000  00000000  00123524  2**2
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 19 .debug_loclists 00018013  00000000  00000000  0012514c  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 20 .debug_rnglists 00001126  00000000  00000000  0013d15f  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS

```

closed 2024-10-31 by prismv

c3d, 2024-10-31:
This may simply be because I switched back to `-Os` for the DM42 builds around September 15, in git commit 3b176a4ac164a719e8890277f08470080af2dd70

prismv, 2024-10-31:
Thanks for your hint. This explains the changed behaviour better than any minor updates of the gcc compilers on my PC.