c3d/db48x #1081: Help viewer gets stuck on Path not found error

mahi2003, 2024-07-30:
I know the images did not ship with 0.7.12 and that a "Path not found" error is normal in the help viewer, but under certain circumstances it appears as if the calculator ends up in an error loop.

The steps below require DB48X without the images present:

* Open the Mathematics Constants menu
* Press and hold `π` to display the help for pi
* Press right/down → "Path not found" error (= normal) 
* Continue pressing right/down exactly 10 more times → "Path not found" error (now for the missing image for Euler's constant)
* Press right/down, multiple times if desired → :x: The help viewer does not scroll and the "Path not found" error persists
* Press left/up, multiple times if desired → :x: The help viewer does not scroll and the "Path not found" error persists
* Press `EXIT`, multiple times if desired → :x: The error persists and the help viewer is not closed

It's as if the calculator is frozen/stuck in the help viewer, but you can escape this situation by pressing the `Home` soft key.

c3d, 2024-07-30:
The "path not found" error persists while there is an image potentially being displayed on the screen.

As you discovered yourself, the `Home` softkey fixes it because it brings the help viewer at a position where there is no image being shown.

The non-intuitive aspect of it is that `EXIT` does not get you out, because it clears the error instead of clearing the help as usual, and then the same help displays again with the same error. Hence the impression that it's stuck.

As for shipping the images, thanks for the reminder. I think that the next release will only ship the `tar.gz` file.

c3d, 2024-07-30:
Marking it as low-priority because it's a transient state of affairs that should be fixed soon ;-)