c3d/db48x #1102: Enter and evaluating π

prismv, 2024-08-10:
π is only visible as an f-shifted alpha key. Is there any faster way to insert it?

Furthermore 1_π or π does not evaluate to a floating number by `→Num`, whereas π entered by f-shift SPC on emu48 and `→Num` indeed results in 3.1415926... (I'm using the standard units table)

Although I memorize a few digits, if I want to use the full precision, currently I have to use - as an HP48/50 and db48x dummy - something like 180 (in DEG mode) `→Rad` and `Value`.

c3d, 2024-08-11:
Hi @prismv,

The π that you insert as an shifted alpha key is only a symbol name. Whether this (as well as `pi`) should identify an RPL command is still an open question, and various releases have been going back and forth on this specific topic. The reason for that debate is that it would be inconsistent to treat π differently from other constants, see below. At least for now, that is the reason `π` will not evaluate numerically with `→Num`: it's a symbol and not a constant.

DB48X has an extensive catalog of constants, which are under the `CONST` menu (shifted I, mnemonically labeled π on the original DM42 keyboard). Under the `Mathematics` submenu, you will find π, e, i, j, ∞ and ? (undefined result).
For each constant, you can either use the symbolic value (which shows up in bold in equations to distinguish it from user symbols) or the numerical value by shifting it. There are also other constant categories, like chemistry or physics, with constants such as *c* (the speed of light).

It is normal for `1_π` to not evaluate to a floating-point number, since it is already a floating-point number with a unit attached. DB48X, like the HP48, lets you use custom units.

Finally, it is actually a nice demo of DB48X capabilities that it can _compute_ π with a large number of digits using a very small number of keystrokes. For example, type the following key sequence:

    shift-O     @ DISP
    200 F5      @ SIG
    200 F6      @ PREC
    shift-N     @ MODES
    F1          @ RAD
    1 shift-L   @ TAN⁻¹
    4 *         @ compute 4*atan(1)=π with 200 digits

c3d, 2024-08-11:
Closing as "won't fix" for now.
@prismv, please reopen if my answers are not satisfactory.

closed 2024-08-11 by c3d

prismv, 2024-08-11:
Will try your example. Thank you for the great response. Now I know better.