c3d/db48x #1103: Result of StandardDeviation is wrong

prismv, 2024-08-10:
I recognize in the code that `StandardDeviation` and `PopulationStandardDeviation` are available and the respective variances.

I guess the former is the sample std.dev. (using dof=n-1) and the latter uses n as denominator for variance calculation.

Inserting sum data vector [1 2 6] in STAT gives the right total of 9 and mean of 3, and should give a standard deviation of (evaluated) `sqrt(7)`, i.e. 2.64575..., but the result is 2.3452... 

Just for completeness the population std.dev. should be (evaluated) `sqrt(4 2/3)`, i.e. 2.1602...

c3d, 2024-08-13:
Good catch, @prismv, thanks a lot for reporting. Will be fixed in 0.7.14.

prismv, 2024-08-13:
De rien. Thnx for the fast fix!

commit 9419c697 (2024-08-13): stats: When computing a sum, evaluate the expression on all terms.

The `total` internal function performs a sum of terms.
The code was optimizing away the sum to zero for the first term, but
as a result ended up not evaluating the term. So the first term in the
sum for the variance was evaluated as `d` instread if `(d-avg)^2`.

Fixes: #1103
commit fe295d03 (2024-08-16): stats: When computing a sum, evaluate the expression on all terms.

The `total` internal function performs a sum of terms.
The code was optimizing away the sum to zero for the first term, but
as a result ended up not evaluating the term. So the first term in the
sum for the variance was evaluated as `d` instread if `(d-avg)^2`.

Fixes: #1103
closed 2024-08-18 by c3d